zephyrproject-rtos/zephyr #23110: PTS: Bluetooth: GATT/SR/GAS/BV-07-C

agata-ponitka, 2020-02-26:
Describe the bug:
After re-connection and second read request, data base is out of sync but IUT returned valid value. 

Expected behavior
IUT returned error.  

[GATT_SR_GAS_BV_07_C_2020_02_26_16_14_15.zip](https://github.com/zephyrproject-rtos/zephyr/files/4256532/GATT_SR_GAS_BV_07_C_2020_02_26_16_14_15.zip)

aescolar, 2020-02-26:
CC @jhedberg

Vudentz, 2020-02-26:
Are there any logs from Zephyr, it might be clearing in case of:

	/* BLUETOOTH CORE SPECIFICATION Version 5.1 | Vol 3, Part G page 2347:
	 * 2.5.2.1 Robust Caching
	 * A connected client becomes change-aware when...
	 * The server sends the client a response with the error code set to
	 * Database Out Of Sync and then the server receives another ATT
	 * request from the client.
	 */

It weird since we do have 2 flags exactly to track this state properly, or does PTS expects that we clearn out of sync state when disconnected, so it only when 2 consecutives ATT request happen within the same connection that would make the client change aware? It would be good to have what the test expection as well.

carlescufi, 2020-02-26:
@Vudentz 
```
4.8.10 GATT/SR/GAS/BV-07-C [Handling Client Requests after a Hash Read or
Service Changed Indication – Database Changed between Connections]
• Test Purpose
Verify that the IUT acting as GATT Server does not return the Database Out of Sync error after a
hash read when the Robust Caching feature is enabled. The database is changed between
connections after a bond has been created.
• References
[11] 2.5.2, 7.2
• Initial Conditions
The GATT database on the IUT contains at least one characteristic that supports the GATT
Characteristic Value Read sub-procedure. Any change on the GATT database during the test
procedure will not affect this characteristic.
The IUT and the Lower Tester have bonded and the connection has been terminated.
The Lower Tester has discovered the characteristics on the IUT and has saved the handles of their
values.
The Lower Tester has enabled the Robust Caching feature.
The Lower Tester has not enabled indications on the Service Changed characteristic.
Test Procedure
1. The Upper Tester orders the IUT to change its GATT database, by adding or removing services and/or characteristics, then the Lower Tester creates a connection with the IUT.
2. The Lower Tester executes the GATT Read Using Characteristic UUID sub-procedure on the Database Hash characteristic.
3. The Lower Tester executes the GATT Characteristic Value Read sub-procedure on another characteristic (different than the Database Hash characteristic) and expects the IUT to return either the characteristic value or a different error code. If the Database Out of Sync error is returned, the test ends with a Fail Verdict.
4. The Lower Tester enables indications on the Service Changed characteristic, then it terminates the connection.
5. Repeat step 1 and wait for the IUT to send an indication on the Service Changed characteristic. If the indication is not received within 30 seconds, the test ends with a Fail Verdict.
6. The Lower Tester confirms the indication, then repeat step 3.
7. The Lower Tester terminates the connection, then reconnects, and then it executes a GATT Characteristic Value Write procedure on the Client Supported Features characteristic, setting the Robust Caching bit to 0, and expects the IUT to return the Value Not Allowed error.
8. The Lower Tester disables indications on the Service Changed characteristic, then it terminates the connection, then repeat step 1, and then the Lower Tester executes the GATT Characteristic Value Read sub-procedure on another characteristic (different than the Database Hash characteristic) and expects the IUT to return the Database Out of Sync error.
9. The Lower Tester terminates the connection, then reconnects, and then it executes the GATT Characteristic Value Read sub-procedure on another characteristic (different than the Database Hash characteristic) and expects the IUT to return the Database Out of Sync error.

Expected Outcome
Pass Verdict
In both executions of step 3, the IUT does not return the Database Out of Sync error.
In step 7, the IUT returns the Value Not Allowed error.
In steps 8 and 9, the IUT returns the Database Out Of Sync error.
```

carlescufi, 2020-02-26:
@Vudentz which logs do you want on the zephyr side?

PR #23116 (2020-02-27): Bluetooth: GATT: Fix not clearing CF_OUT_OF_SYNC
If the client is change-unware and disconnects the spec requires that
the stack still sends the error out of sync for the next request:

'The ATT_ERROR_RSP PDU is sent only once after the client becomes
change-unaware, unless the client _disconnects_ or the database changes
again before the client becomes change-aware in which case the
ATT_ERROR_RSP PDU shall be sent again'

Fixes #23110
commit 9e8b78b8 (2020-02-27): Bluetooth: GATT: Fix not clearing CF_OUT_OF_SYNC

If the client is change-unware and disconnects the spec requires that
the stack still sends the error out of sync for the next request:

'The ATT_ERROR_RSP PDU is sent only once after the client becomes
change-unaware, unless the client _disconnects_ or the database changes
again before the client becomes change-aware in which case the
ATT_ERROR_RSP PDU shall be sent again'

Fixes #23110
closed 2020-02-27 by jhedberg