zephyrproject-rtos/zephyr #23111: drivers:usb:device:sam0:  Descriptor tables are filled with zeros in attach()

sslupsky, 2020-02-26:
**Describe the bug**
During review of the sam0 usb device driver, a question comes to mind regarding the descriptor tables.

When usb_dc_attach() is called, the descriptor tables are filled with zeros.  At first glance, this appears to be the correct thing to do.  Upon a deeper investigation however, I am not so sure.  The descriptor tables are initialized in usb_device_init() which then "fixes" the descriptor table by calling usb_fix_descriptor() which appears to modify the descriptor.

Thus it seems apparent that an attach() subsequently "undoes" the "fix" by filling the entire descriptor with zeros.  

It would seem that the "fix" would also need to be done again in the attach()?

jfischer-phytec-iot, 2020-02-28:
> During review of the sam0 usb device driver, a question comes to mind regarding the descriptor tables.

Then it is probably a question and it would be properly to label it as a question?
You mean the "Endpoint descriptors" table?

> 
> When usb_dc_attach() is called, the descriptor tables are filled with zeros. At first glance, this appears to be the correct thing to do. Upon a deeper investigation however, I am not so sure. The 

The ["_descriptor tables filled with zeros_"](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_sam0.c#L232) are internal Endpoint Descriptor structure for the SAM's device controller, used to by a built-in DMA. (remark: Kinetis USB IP has similar descriptor table structure. This buffer descriptor can also help to reduce the number of endpoint buffer copies for example, but it currently not used in device stack.)

>descriptor tables are initialized in usb_device_init() which then "fixes" the descriptor table by calling usb_fix_descriptor() which appears to modify the descriptor.

No, usb_fix_descriptor() is USB device stack internal function and is used to "fix" interface/endpoint descriptors because they to not know from each other as a independent part of USB device descriptor. It has nothing to do with endpoint-buffer descriptor table from the specific USB device controller.

> 
> Thus it seems apparent that an attach() subsequently "undoes" the "fix" by filling the entire descriptor with zeros.
> 
> It would seem that the "fix" would also need to be done again in the attach()?

sslupsky, 2020-02-29:
@jfischer-phytec-iot You are of course correctand I apologize, when i posted the isssue there are only four "choices" for the type of issue and the closest I could come to "question" was unfortunately "bug" since the other choices were even more inappropriate.

I am not familiar with the Kinetis devices so cannot comment on those.  The sam0 usb peripheral appears to me to not include hardware buffers but rather it looks like it uses hardware pointers to buffers/tables.  These buffers/tables are allocated in the mcu sram?

jfischer-phytec-iot, 2020-03-04:
> The sam0 usb peripheral appears to me to not include hardware buffers but rather it looks like it uses hardware pointers to buffers/tables. These buffers/tables are allocated in the mcu sram?

Yes, both [Endpoint Descriptors](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_sam0.c#L28) and [Endpoint Buffers (u32_t type!)](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_sam0.c#L37) are located in the RAM. During attach the Descriptor Address Register [DESCADD](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_sam0.c#L233) will be initialized to the address of Endpoint Descriptor structure. Specific Endpoint Buffer will be assigned during `usb_dc_ep_configure()` to the [ADDR](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/usb/device/usb_dc_sam0.c#L381) property of the Endpoint Descriptor.

The driver does not use double buffering (Ping-Pong Operation). This is a drawback with Zephyr's USB stack buffer handling (too much copies), and can lead to the lost of the ACK packet from the host during control IN stage (can be reproduced using testusb sample).

sslupsky, 2020-03-06:
Thanks for the feedback.  Regarding the sam0 driver, the descriptor buffer is filled with zeros:
https://github.com/zephyrproject-rtos/zephyr/blob/6f3f3c230cf4384e3f5f35ebfdf6b5e84db38997/drivers/usb/device/usb_dc_sam0.c#L232
and I was wondering if might cause some kind of problem because set_config() is called during the device_init() before the dc_attach().  If set_config() updates the descriptor (via fix_descriptor()?), are those changes lost when a subsequent attach() is performed?

jfischer-phytec-iot, 2020-03-10:
> I was wondering if that might cause some kind of problem because usb_get_device_descriptor() is called during the device_init() before the dc_attach(). If usb_get_device_descriptor() updates the descriptor (via fix_descriptor()?), are those changes lost when a subsequent attach() is performed?

You still mix it up. The Endpoint Buffer Descriptor from usb_dc_sam0.c has nothing to do with the  USB Endpoint Descriptor, which is a part of USB Interface Descriptor and so on.

>Regarding the sam0 driver, the descriptor buffer is filled with zeros in dc_attach():

Correct is: the **SAM0 specific** **_Endpoint Buffer Descriptor_** is filled with zeros in dc_attach()

sslupsky, 2020-03-14:
Not sure if I am following you here.  It appears you agree with my report that the **sam0** endpoint buffer descriptor is being filled with zeros?

So, is this something that should be fixed?

jfischer-phytec-iot, 2020-03-16:
> Not sure if I am following you here.

I am not sure even. Do you agree that **SAM0 specific Endpoint Buffer Descriptor** has noting to do with **USB descriptors**?

>It appears you agree with my report that the **sam0** endpoint buffer descriptor is being filled with zeros?

I cannot not agree as it is the fact.

> 
> So, is this something that should be fixed?

I think not, or fixed by what?
Do you have attach/detach issues with usb_dc_sam0? For example, `brk` property from `struct_sam0_data` increments in `usb_dc_ep_configure()` but never decrements even not in `usb_dc_detach()` or `usb_dc_ep_disable()`. There are few bug/issues regarding endpoint buffer handling related to detach/attach and usb_dc_ep_configure/usb_dc_ep_disable.

sslupsky, 2020-03-16:
Yes, detach and attach issues.  I can attach / detach three times and then it crashes.

Ahh, I see the brk issue.  This might be the cause of my problem.  I'll have a closer look.

Thank you.

carlescufi, 2020-05-20:
@sslupsky can this be closed now?

sslupsky, 2020-05-20:
@carlescufi I haven't had a chance yet to look into the brk issue @jfischer-phytec-iot mentioned.  I'll try to do that next week.

jfischer-phytec-iot, 2020-06-25:
> Do you have attach/detach issues with usb_dc_sam0? For example, brk property from struct_sam0_data increments in usb_dc_ep_configure() but never decrements even not in usb_dc_detach() or [...]

#26288 should resolve this issue.

closed 2020-07-06 by nashif

sslupsky, 2022-08-17:
@carlescufi I finally had a chance to rebase and test again.  The issue is still unresolved.  However, I did discover the underlying cause of the crash which is a memory leak.  I've submitted a PR #49190

PR #26288 (2020-07-06): drivers: usb: switch the SAM0 driver from a custom allocator to the heap
Also automatically enable the heap if the USB device is selected.

Part of #23178

Fixes #23178
Fixes #23111