zephyrproject-rtos/zephyr #23117: Unable to flash HelloWorld w/XDS-110 & OpenOCD

phuntimes, 2020-02-26:
**Describe the bug**
Flashing failed while attempting to follow getting started guide for LAUNCHXL-CC2640R2F. OpenOCD fails to negotiate with XDS-110. Made sure XDS-110 has latest firmware via CCS (see output). Tried workaround suggested [here](https://github.com/zephyrproject-rtos/zephyr/issues/21372#issuecomment-574897770).

**To Reproduce**
Steps to reproduce the behavior:
1. Follow [Getting Started guide](https://docs.zephyrproject.org/latest/getting_started/index.html) with build option `-b cc26x2r1_launchxl`
2. `west build`

**Expected behavior**
As stated in Getting Started guide.

**Impact**
 - showstopper

**Screenshots or console output**
Using (default) 2500kHz:
```
(zephyr) user@LOC-USER zephyr % west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01089-g3bfe4926 (2020-02-26-11:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter srst delay: 100

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.7
Info : XDS110: hardware version = 0x002b
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 2500 kHz
Info : TAP cc26x2.jrc does not have valid IDCODE (idcode=0x7733405e)
Info : JTAG tap: auto0.tap tap/device found: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Error: cc26x2.jrc: IR capture error; saw 0x02 not 0x01
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Info : TAP cc26x2.jrc does not have valid IDCODE (idcode=0x7733405e)
Info : JTAG tap: auto0.tap tap/device found: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Error: cc26x2.jrc: IR capture error; saw 0x02 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
FATAL ERROR: command exited with status 1: /usr/local/bin/openocd -s [$ZEPHYRPROJECT]/zephyr/boards/arm/cc26x2r1_launchxl/support -f [$ZEPHYRPROJECT]/zephyr/boards/arm/cc26x2r1_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
Modifying `board/ti_cc26x2_launchpad.cfg`:
```
(zephyr) user@LOC-USER zephyr % west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner openocd
-- runners.openocd: Flashing file: [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex
Open On-Chip Debugger 0.10.0+dev-01089-g3bfe4926 (2020-02-26-11:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter srst delay: 100

Info : XDS110: connected
Info : XDS110: firmware version = 3.0.0.7
Info : XDS110: hardware version = 0x002b
Info : XDS110: connected to target via JTAG
Info : XDS110: TCK set to 2500 kHz
Info : clock speed 1500 kHz
Info : JTAG tap: cc26x2.jrc tap/device found: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Warn : JTAG tap: cc26x2.jrc       UNEXPECTED: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Error: JTAG tap: cc26x2.jrc  expected 1 of 1: 0x0bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0* cc26x2.cpu         cortex_m   little cc26x2.cpu         tap-disabled

Info : JTAG tap: cc26x2.jrc tap/device found: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Warn : JTAG tap: cc26x2.jrc       UNEXPECTED: 0xbb99a02f (mfg: 0x017 (Texas Instruments), part: 0xb99a, ver: 0xb)
Error: JTAG tap: cc26x2.jrc  expected 1 of 1: 0x0bb4102f (mfg: 0x017 (Texas Instruments), part: 0xbb41, ver: 0x0)
Error: Trying to use configured scan chain anyway...
Warn : Bypassing JTAG setup events due to errors
Error: Target not examined yet
Error: auto_probe failed
auto erase enabled
Info : XDS110: disconnected
FATAL ERROR: command exited with status 1: /usr/local/bin/openocd -s [$ZEPHYRPROJECT]/zephyr/boards/arm/cc26x2r1_launchxl/support -f [$ZEPHYRPROJECT]/zephyr/boards/arm/cc26x2r1_launchxl/support/openocd.cfg -c init -c targets -c 'reset halt' -c 'flash write_image erase [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex' -c 'reset halt' -c 'verify_image [$ZEPHYRPROJECT]/zephyr/build/zephyr/zephyr.hex' -c 'reset run' -c shutdown
```
**Environment (please complete the following information):**
 - OS: 
    - macOS 110.15.3 (19D76)
    - python 3.8.1
 - Toolchain:
    - west (v0.7.2)
    - gnuarmemb (9-2019-q4-major)
    - cmake (3.16.4)
    - ninja (1.10.0)
    - openocd (0.10.0)
 - Commit SHA or Version used
    - master (a410486dcad6f83e49db6d26d9d31d5d632917f3)

**Additional context**
 - openocd built from source from sourceforge repo using default configure
 - pyenv used for virtual environment for west and dependencies

phuntimes, 2020-02-26:
I realized I may have made a mistake here. What I had on hand to test with was a CC2640 (M3), not a CC2642 (M4).  I'm going to not close this for the moment b/c I want to see if this is an openocd and not the MCU.

vanti, 2020-03-03:
@phuntimes The CC2640R2F belongs to a different family of devices from the CC26x2, so I wouldn't have expected this to work.

nashif, 2020-03-11:
based on comment from @vanti, not a bug.

closed 2020-03-11 by nashif