zephyrproject-rtos/zephyr #23133: boards: adafruit_feather_m0: don't throw compiler warnings on using of sercom1-3

KubaFYI, 2020-02-27:
In order to use it's serial peripheral SERCOMs SAD1 requires their pins to be configured with pinmux. Currently that's enabled only for the board default serial interfaces: the USART TX/RX on sercom0, an SPI interface on sercom4, and the USB port. The pinmux code currently is written to throw a compiler error if a user app tries to use sercom1-sercom3, claiming that the pin mapping have not been configured but at the same time it doesn't give a chance for this configuration to take place in app code.

I propose that those errors are changed to warnings which would allow users to proceed with compilation and configure the pin mappings themselves should they wish to use sercom1-sercom3.

KubaFYI, 2020-02-27:
What do you think @henrikbrixandersen ?

henrikbrixandersen, 2020-03-02:
Sounds like an okay solution to me.

PR #23252 (2020-04-17): boards: pinmux: Don't throw compiler warnings on using custom sercom use
In order to use it's serial peripheral SERCOMs SAMD socs require their
pins to be configured with pinmux. Currently pins are muxed only for the
boards' default serial interfaces. The pinmux code is written to throw
a compiler error if a user tries to use any sercom in a way it wasn't
pre-defined to be used.

This commit changes compiler errors to warnings so that user can provide
custom pinmuxing code in their app.

Fixes #23133
commit 3523c1b8 (2020-04-17): boards: pinmux: Don't throw compiler warnings on using custom sercom use

In order to use its serial peripheral SERCOMs SAMD socs require their
pins to be configured with pinmux. Currently pins are muxed only for the
boards' default serial interfaces. The pinmux code is written to throw
a compiler error if a user tries to use any sercom in a way it wasn't
pre-defined to be used.

This commit changes compiler errors to warnings so that user can provide
custom pinmuxing code in their app.

Fixes #23133
closed 2020-04-17 by galak