zephyrproject-rtos/zephyr #23139: USB Mass storage - Unexpected USB restart from host

optdcw, 2020-02-27:
I have created appliaction which uses internal flash of nrf52840. 
I has size 400Kb. 
I'm using modified usb mass storage. It works as expected with RAM FAT. It also works with small files inside the internal flash.
However, when i try to copy larger file around 300Kb the transfer is slow which is expected however around 100Kb the usb device recieves USB reset from the host device(Windows/Linux).

I expect that the SCSI protocol implemented inside the usb_mass storage does not have implemented some kind of feed back to the host, because the host does not even show progress of the copy.

I'm could able to modify the usb_mass storage to work properly if somebody could give me some hints regarding SCSI protocol implemeted inside. I have studied the usb bulk transfer which is used here, however i could not figure out what is the true reason for the host reset.

The log from the device is attached from bellow.
```
[00:01:18.716,918] [0m<dbg> usb_msc.memoryWrite: Disk WRITE Qd 164[0m
[00:01:18.716,949] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.716,979] [0m<dbg> usb_msc.mass_storage_bulk_out: > BO not clearing NAKs yet[0m
[00:01:18.717,010] [0m<dbg> usb_msc.mass_thread_main: sem 3[0m
[00:01:18.865,753] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.865,783] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.865,936] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.865,997] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.866,119] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.866,241] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.866,333] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.866,424] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.866,516] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.866,607] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.866,699] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.866,821] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.866,912] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.867,004] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.867,095] [1;31m<err> usb_nrfx: read complete, ep 0x02, len 64[0m
[00:01:18.867,156] [0m<dbg> usb_msc.memoryWrite: Disk WRITE Qd 165[0m
[00:01:18.867,156] [0m<dbg> usb_msc.mass_storage_bulk_out: > BO not clearing NAKs yet[0m
[00:01:18.867,218] [1;31m<err> usb_nrfx: read request, ep 0x02[0m
[00:01:18.867,218] [0m<dbg> usb_msc.mass_thread_main: sem 3[0m
[00:01:18.952,331] [1;31m<err> usb_nrfx: USBD reset event[0m
[00:01:18.952,453] [0m<dbg> usb_msc.mass_storage_status_cb: USB device reset detected[0m
[00:01:18.959,381] [1;31m<err> usb_nrfx: SETUP: r:6 rt:128 v:256 i:0 l:64[0m
[00:01:18.959,381] [1;31m<err> usb_nrfx: ep_read: ep 0x00, maxlen 8[0m
[00:01:18.959,472] [1;31m<err> usb_nrfx: ep_write: ep 0x80, len 18[0m
[00:01:18.959,564] [1;31m<err> usb_nrfx: ctrl write complete[0m
[00:01:18.959,716] [1;31m<err> usb_nrfx: USBD reset event[0m
[00:01:18.959,747] [0m<dbg> usb_msc.mass_storage_status_cb: USB device reset detected[0m
[00:01:19.023,468] [1;31m<err> usb_nrfx: ep_write: ep 0x83, len 13[0m
[00:01:19.057,220] [1;31m<err> usb_nrfx: SETUP: r:6 rt:128 v:256 i:0 l:18[0m
[00:01:19.057,220] [1;31m<err> usb_nrfx: ep_read: ep 0x00, maxlen 8[0m
[00:01:19.057,250] [1;31m<err> usb_nrfx: ep_write: ep 0x80, len 18[0m
[00:01:19.057,342] [1;31m<err> usb_nrfx: ctrl write complete[0m
[00:01:19.057,495] [1;31m<err> usb_nrfx: SETUP: r:9 rt:0 v:1 i:0 l:0[0m
[00:01:19.057,495] [1;31m<err> usb_nrfx: ep_read: ep 0x00, maxlen 8[0m
[00:01:19.057,556] [1;33m<wrn> usb_device: Failed to enable endpoint 81[0m
[00:01:19.057,556] [1;33m<wrn> usb_device: Failed to enable endpoint 82[0m
[00:01:19.057,556] [1;33m<wrn> usb_device: Failed to enable endpoint 1[0m
[00:01:19.057,586] [1;33m<wrn> usb_device: Failed to enable endpoint 83[0m
[00:01:19.057,586] [1;33m<wrn> usb_device: Failed to enable endpoint 2[0m
[00:01:19.057,617] [1;33m<wrn> usb_device: Failed to enable endpoint 84[0m
[00:01:19.057,617] [1;33m<wrn> usb_device: Failed to enable endpoint 85[0m
[00:01:19.057,647] [1;33m<wrn> usb_device: Failed to enable endpoint 3[0m
[00:01:19.057,678] [0m<inf> usb_cdc_acm: USB device configured[0m
[00:01:19.057,678] [0m<inf> usb_cdc_acm: USB device configured[0m
[00:01:19.057,678] [0m<dbg> usb_msc.mass_storage_status_cb: USB device configured[0m
[00:01:19.057,708] [1;31m<err> usb_nrfx: ep_write: ep 0x80, len 0[0m
[00:01:59.728,820] [1;31m<err> usb_nrfx: USBD reset event[0m
[00:01:59.728,881] [0m<dbg> usb_msc.mass_storage_status_cb: USB device reset detected[0m
```

jfischer-phytec-iot, 2020-02-28:
> I have created appliction which uses internal flash of nrf52840 as USB Mass storage device.

I guess the host resets the device because erase/write flash takes too long and during that the device can not respond. For example USB DFU selects IMG_ERASE_PROGRESSIVELY option to prevent this behavior. So the issue is in your application, I will replace the bug label with question for now. Note Mass Storage Class has a issue tracked here #14302, which will probably not be fixed in this release.

optdcw, 2020-02-28:
I have suspected this kind of behavior. The problem is that the SCSI command issued by the host is single large bulk transaction. 
I think solution to this could be implementation some kind of SCSI command which tells the host device how to limit the maximum length of the SCSI transaction. 

I found stackoverflow thread mentioning solution which could work:
https://stackoverflow.com/questions/33354858/how-can-i-limit-the-number-of-blocks-written-in-a-write-10-command
It talks about "Block Limits VPD page", which can be implemented by the device to tell the host how to limit the SCSI transaction, however i would be greatfull if somebody could confirm my suspision, before i implement it.

carlescufi, 2020-03-04:
CC @nordic-auko. Do you have additional info for @optdcw?

optdcw, 2020-03-04:
I have implementace the support for the INQUIRY command and the VPD pages. 
More specificly the mandatory the VPD support page and optional Block limits VPD page.

I found out that the os Linux and windows ignores these pages for usb mass storage device unfortunatly.

On the linux i was able to change the max transaction size from the os. However that is not usable solution for non development scenarios.

Currently i dont know how to proceed next. To fix this issue.

jfischer-phytec-iot, 2020-03-04:
I still do not see that we have to implement something in Mass Storage Class and it does not seems like you have a plan.

On nRF52804 `The flash is divided into 256 pages of 4 kB`, Time to erase one page = 85ms, Time to write one 32-bit word = 41µs, there is also partial erase support (as I mentioned above)

The block size of MSC is fixed to 512 bytes, this is a little misaligned to flash page size. There are a few systems between USB und nRF flash driver, not easy to evaluate how it behaves as whole.

optical-o, 2020-03-04:
The problem is that the host system issues single large bulk transaction which blocks the USB responses.  The inner SCSI protocol requires to write whole block to the device in this case 64Kb(default value on windows).

The partial erasing is done within the inner SCSI transaction, however the timeout is on the outer layer USB bulk transfer.

I think only solution is to persuade the host system to issue smaller SCSI transaction or to optimize the usb mass storage to be able to write blocks of 64Kb faster then is the USB timeout value. 

My last attempt with implementing SCSI compliant solution using the VPD pages failed.

This leaves only the option to optimize the USB mass storage driver to be faster.

jfischer-phytec-iot, 2020-03-04:
>  or to optimize the usb mass storage to be able to write blocks of 64Kb faster then is the USB timeout value.

We can discuss it for long time, probably without any conclusion. Again: Block size in MSC is fixed to 512 bytes, the host should not transmit larger blocks, USB FS bulk EP buffer is 64 bytes, there is nothing more to optimize. MSC is just a adaption layer between USB and disk subsystem. We had a issue in USB DFU because of timeout during flash erase, and that was one of the reasons to add 
IMG_ERASE_PROGRESSIVELY option. (@nvlsianpu, FYI, you can probably describe it better)

nordic-auko, 2020-03-04:
I did a test on an nRF52840, writing a 400 KiB file to the internal flash via USB MSC. If I do the disk writing directly from the bulk_out handler instead of offloading it to the op thread, the write operation succeeds without a reset or other errors. It does take a long time to complete though. Note the timestamps in the USB sniffer screenshot below:

[image]

jfischer-phytec-iot, 2020-03-04:
> If I do the disk writing directly from the bulk_out handler instead of offloading it to the op thread, the write operation succeeds without a reset or other errors. It does take a long time to complete though.

Can you share the patch? I would expect reverse behavior. 
 
> (the Host (win10) does write the file in 64 KiB increments)

But MSC passes it as max. 512 byte blocks to `disk_acces_write()`.

nordic-auko, 2020-03-04:
> Can you share the patch? I would expect reverse behavior.

Copied the actions from THREAD_OP_WRITE_QUEUED into the bulk_out handler:

```
diff --git a/subsys/usb/class/mass_storage.c b/subsys/usb/class/mass_storage.c
index 08cdadf3b8..0ba853c0cd 100644
--- a/subsys/usb/class/mass_storage.c
+++ b/subsys/usb/class/mass_storage.c
@@ -654,7 +654,26 @@ static void memoryWrite(u8_t *buf, u16_t size)
 			LOG_DBG("Disk WRITE Qd %d", (addr/BLOCK_SIZE));
 			thread_op = THREAD_OP_WRITE_QUEUED;  /* write_queued */
 			defered_wr_sz = size;
-			k_sem_give(&disk_wait_sem);
+			// k_sem_give(&disk_wait_sem);
+			if (disk_access_write(disk_pdrv,
+						page, (addr/BLOCK_SIZE), 1)) {
+				LOG_ERR("!!!!! Disk Write Error %d !!!!!",
+					addr/BLOCK_SIZE);
+			}
+
+			addr += size;
+			length -= size;
+			csw.DataResidue -= size;
+
+
+			if ((!length) || (stage != MSC_PROCESS_CBW)) {
+				csw.Status = (stage == MSC_ERROR) ? CSW_FAILED : CSW_PASSED;
+				sendCSW();
+			}
+
+			thread_op = THREAD_OP_WRITE_DONE;
+
+			usb_ep_read_continue(mass_ep_data[MSD_OUT_EP_IDX].ep_addr);
 			return;
 		}
 	}

```

nvlsianpu, 2020-03-05:
I'm sure that if nRF have to erase 64 KB in one transaction it will timeout the USB bus (as NVMC will lock any code execution). I'm not sure what happens here - will have time to investigate maybe next week. One flash pager erase will take ~89 ms, Probably need to synchronize this somehow with the USB machinery.

carlescufi, 2020-05-13:
@optdcw @nordic-auko could you please retest this with the latest master or 2.3.0-rc1 to see if it's still an issue.

nordic-auko, 2020-05-25:
> 
> 
> @optdcw @nordic-auko could you please retest this with the latest master or 2.3.0-rc1 to see if it's still an issue.

Seems to work here. Successfully copied a 400 KiB file to the nRF52840 FAT flash disk.

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi