zephyrproject-rtos/zephyr #23149: tests/drivers/watchdog/wdt_basic_api failed on sam_e70_xplained board with v1.14 branch.

chen-png, 2020-02-28:
**Describe the bug**
Running tests/drivers/watchdog/wdt_basic_api test case on sam_e70_xplained board with v1.14 branch, it will loop to report the message below.

This test has been skipped on sam_e70_xplained board in master branch.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=sam_e70_xplained  tests/drivers/watchdog/wdt_basic_api 
2. west flash
3. See error

**Screenshots or console output**
```
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
Running test suite wdt_basic_test

starting test - test_wdt
Testcase: test_wdt_no_callback
Waiting to restart MCU
```

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: ff682926

chen-png, 2020-08-06:
it's still an issue, reopened it

carlescufi, 2020-09-08:
We are going to fix this for 1.14. If the bug exists in master, we will fix it there.

closed 2020-09-08 by carlescufi