zephyrproject-rtos/zephyr #23156: App determines if Bluetooth host link request is allowed

LuoZhongYao, 2020-02-28:
Applications sometimes need to make some restrictions on Bluetooth link requests.For example, in undiscoverable mode, only devices in the pairing list are allowed to link, or only certain types of devices with specific addresses are allowed to link.

**Describe the solution you'd like**
Can add a callback function to bt_conn_cb. This callback is called when a connection request is received. If the callback returns true, then the link is allowed, otherwise the link is rejected. Multiple return values ​​are and relations

aescolar, 2020-02-28:
CC @jhedberg @joerchan @carlescufi @Vudentz

jhedberg, 2020-02-28:
"callback is called when a connection request is received" - what exactly do you mean by "connection request" @LuoZhongYao ? BR/EDR has a separate connection request HCI event, however LE doesn't, i.e. the controller just informs the host of a new connection using the LE Connection Complete event, without the host having a chance to reject this in advance.

LuoZhongYao, 2020-02-29:
> "callback is called when a connection request is received" - what exactly do you mean by "connection request" @LuoZhongYao ? BR/EDR has a separate connection request HCI event, however LE doesn't, [...]

 connection request HCI event

joerchan, 2020-03-02:
Sounds like a whitelist feature. For LE we have whitelist feature and whitelisting is done in the controller. So with BR/EDR we might want to have a host or application whitelist feature. With the callback suggestion it would be application that whitelists.

@jhedberg Maybe just add a BR/EDR-only connection request callback as suggested?

@LuoZhongYao Is this something you could send a PR for perhaps?

jhedberg, 2020-03-02:
> @jhedberg Maybe just add a BR/EDR-only connection request callback as suggested?

Yes, for BR/EDR that would probably make sense. One open question is whether it should be synchronous (return value indicates accept/reject) or asynchronous (separate API to accept/reject a pending connection)

joerchan, 2020-03-02:
As the default value for the Connection_Accept_Timeout value is 5.06 seconds. I don't see a reason why it shouldn't be made asynchronous.
I don't know all the criteria for accepting the connections. It could be whitelist/blacklist, user-confirmation perhaps? With asynchronous we would cover more use-cases I think.

carlescufi, 2020-04-07:
@LuoZhongYao we don't currently plan to address this. If you'd like to see this in the tree, please send a Pull Request.

closed 2020-04-07 by carlescufi