zephyrproject-rtos/zephyr #23161: I2C and sensor deinitialization

Mhrooczny, 2020-02-28:
Hello everyone,

I'm trying to make low power app. There is any way to deinitialization sensor with I2C bus which is using? I'm using nRF52832 soc and BME280 sensor and I tried to do this using Device Power Management API. I pass the BME280 and I2C to `device_get_binding` function. Then used `device_set_power_state` with `DEVICE_PM_LOW_POWER_STATE` argument. Power draw is lot higher then if i keep only BME280 sensor in `device_get_binding` without `device_set_power_state` set LOW_POWER_STATE but this makes 1s period ~mA pick from BME280 sensor or I2C bus.
The return value for `device_set_power_state` is `0` so it should get job done.

carlescufi, 2020-02-28:
@pabigot and @mbolivar please check if this is a duplicate

Mhrooczny, 2020-02-28:
For UART this method like sending PM to low power mode using `device_set_power_state` is working but for I2C and sensor got higher current draw after calling `device_set_power_state`. I already tried with `OFF` state and `FORCE_SUSPEND` / `SUSPEND`.

mbolivar, 2020-02-28:
Sorry, I'm having trouble understanding the report.

Can you please share exactly which devices you passed to `device_set_power_state()` in both scenarios, and what the results were?

Mhrooczny, 2020-03-02:
Yes sure, @mbolivar i'm sorry for not clear problem description.
For every scenario i'm using PM for UART and setting it low power state to get lowest current draw in `k_sleep()` function. 
1. When BME280 sensor is only binded and unused the current draw is as low as in `k_sleep()` with UART in low power but there is high current peak with 1s period which i think is caused by BME280 sensor. In this scenario I2C is not binded it is only in `prj.conf` `CONFIG_I2C=y`
2. BME280 sensor is binded and set to low power using `device_set_power_state()` return value is 0. Current draw is same like in 1. scenario.
3. BME280 sensor and I2C is binded using `device_get_binding` and both devices are set in low power state using `device_get_binding`. Current draw increases dramatically in stable value about 700uA

mbolivar, 2020-03-02:
Hi and thanks.

The `device_get_binding()` function doesn't do anything to the device. It just returns a pointer to it:

https://github.com/zephyrproject-rtos/zephyr/blob/master/kernel/device.c#L66

Whether or not you call `device_get_binding()` does not affect the power drawn by the device. Even though there is "get" in the function's name, it doesn't actually call a probe function in the driver or anything like that.

> BME280 sensor and I2C is binded using device_get_binding() and both devices are set in low power state using device_set_power_state(). Current draw increases dramatically in stable value about 700uA

It seems like the twim driver properly implements the low power callback, so I am not sure why this is happening.

Paging @anangl -- any idea?

anangl, 2020-03-03:
> It seems like the twim driver properly implements the low power callback, so I am not sure why this is happening.
> 
> Paging @anangl -- any idea?

Yes, the implementations of `device_pm_control` function for both TWI and TWIM drivers look fine. But the BME280 sensor driver does not implement its own `device_pm_control` function, so calling `device_set_power_state()` for the sensor results in the default no-op function (just returning 0) being called. Thus, nothing in the driver or sensor configuration changes.
Regarding the scenario 3, my guess is that the sensor triggers an interrupt and activates its interrupt output, what is handled by the GPIO [callback](https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/sensor/bma280/bma280_trigger.c#L68) and leads to `bma280_thread_cb()` being called. This callback in turn tries to perform some I2C operations to service the interrupt, but since the I2C driver is suspended, the operations will fail and the interrupt in the sensor won't get cleared, so the sensor will stay awake and consume power. Does my reasoning make some sense?

mbolivar-nordic, 2020-03-17:
I've switched my work-related account to @mbolivar-nordic (https://github.com/zephyrproject-rtos/zephyr/pull/23530).

gmarull, 2022-11-21:
The description on this PR is outdated, PM APIs have changed significantly. Runtime PM should likely be used to solve this case.

closed 2022-11-21 by gmarull