zephyrproject-rtos/zephyr #23169: "blinky" sample fails to build for BBC MicroBit (DT_ALIAS_LED0_GPIOS_CONTROLLER undefined)

snej, 2020-02-28:
**Describe the bug**
Following along in the Getting Started guide, I tried to build the "blinky" sample for the BBC MicroBit board. It fails due to a C compile error; apparently the constants for the LED0 pin are not defined as expected. The sample does build for the `reel_board`.

**To Reproduce**
Steps to reproduce the behavior:
1. Setup zephyr as per Getting Started Guide, steps 1-5
2. Follow step 6 by entering `west build -p auto -b bbc_microbit samples/basic/blinky`

**Expected behavior**
The build succeeds

**Impact**
To actually run the sample I need to build it for a board I have in my possession.
I would expect a minimal sample like this to build successfully for every supported board, especially since the guide doesn't specify any particular board but implies any will work.
I do own two other supported boards so I'm not dead in the water yet, but this is disappointing.

**Screenshots or console output**
```
[34/122] Building C object CMakeFiles/app.dir/src/main.c.obj
FAILED: CMakeFiles/app.dir/src/main.c.obj 
ccache /Users/snej/Couchbase/Embedded/Toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/arm-none-eabi-gcc -DBUILD_VERSION=v2.2.0-rc2-41-gb96477dd69bb -DKERNEL -DNRF51 -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/include -Izephyr/include/generated -I/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf51 -I/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/ext/hal/cmsis/Core/Include -I/Users/snej/Couchbase/Embedded/zephyrproject/modules/hal/nordic/nrfx -I/Users/snej/Couchbase/Embedded/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/Users/snej/Couchbase/Embedded/zephyrproject/modules/hal/nordic/nrfx/mdk -I/Users/snej/Couchbase/Embedded/zephyrproject/modules/hal/nordic/. -isystem /Users/snej/Couchbase/Embedded/zephyrproject/zephyr/lib/libc/minimal/include -isystem /Users/snej/Couchbase/Embedded/Toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include -isystem /Users/snej/Couchbase/Embedded/Toolchain/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/include-fixed -Os -imacros/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m0 -imacros/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-address-of-packed-member -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky=CMAKE_SOURCE_DIR -fmacro-prefix-map=/Users/snej/Couchbase/Embedded/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/Users/snej/Couchbase/Embedded/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -mabi=aapcs -march=armv6s-m -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/src/main.c.obj -MF CMakeFiles/app.dir/src/main.c.obj.d -o CMakeFiles/app.dir/src/main.c.obj   -c /Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky/src/main.c
/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky/src/main.c: In function 'main':
/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky/src/main.c:24:27: error: 'DT_ALIAS_LED0_GPIOS_CONTROLLER' undeclared (first use in this function); did you mean 'DT_ALIAS_SW0_GPIOS_CONTROLLER'?
   24 |  dev = device_get_binding(DT_ALIAS_LED0_GPIOS_CONTROLLER);
      |                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                           DT_ALIAS_SW0_GPIOS_CONTROLLER
/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky/src/main.c:24:27: note: each undeclared identifier is reported only once for each function it appears in
/Users/snej/Couchbase/Embedded/zephyrproject/zephyr/samples/basic/blinky/src/main.c:29:32: error: 'DT_ALIAS_LED0_GPIOS_PIN' undeclared (first use in this function); did you mean 'DT_ALIAS_SW0_GPIOS_PIN'?
   29 |  ret = gpio_pin_configure(dev, DT_ALIAS_LED0_GPIOS_PIN,
      |                                ^~~~~~~~~~~~~~~~~~~~~~~
      |                                DT_ALIAS_SW0_GPIOS_PIN
[47/122] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/clock_control/nrf_power_clock.c.obj
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/local/bin/cmake --build /Users/snej/Couchbase/Embedded/zephyrproject/zephyr/build
```

**Environment (please complete the following information):**
 - OS: macOS 10.15.3
 - Toolchain: GCC-ARM
 - Commit: b96477dd69bbc6b1830048e8eee59cf62c3684da (HEAD -> master, origin/master)

ioannisg, 2020-03-02:
>I would expect a minimal sample like this to build successfully for every supported board, especially since the guide doesn't specify any particular board but implies any will work.

The following is present in the README:
```
The
sample code is configured to work on boards that have defined the led0
alias in their board devicetree description file. Doing so will generate
these variables:

- DT_ALIAS_LED0_GPIOS_CONTROLLER
- DT_ALIAS_LED0_GPIOS_PIN
```

Additionally, the sample, itself, is filtered-out for boards without gpio-leds, led0
`    filter: dt_compat_enabled_with_alias("gpio-leds", "led0")`

So, strictly speaking, this does not seem to be a bug, IMHO. Sanity check on bbc_microbit is successfully built:
`./scripts/sanitycheck  -p bbc_microbit -b -T `samples/

Since the platform supports LEDs, I think it could run this sample, though, but this is an enhancement, then. @snej  contributions are welcome if you would like to help, getting blinky running on microbit :)

snej, 2020-03-02:
Not a bug in the sample, then, but a bug in the documentation — it would be nice to fix the Getting Started guide so it calls out boards that won't work, or advises you to look at the sample's README (in which case it'd also be nice if the sample mentioned that the MicroBit is not supported.)

> contributions are welcome if you would like to help, getting blinky running on microbit :)

To be frank, at this point I'm still evaluating whether I want to use Zephyr at all, vs. one of the many other RTOSs out there. Every roadblock I hit while going through the friendly startup guide reduces my enthusiasm. I think it's in the Zephyr maintainers' interest to fix these, whether or not they consider them bugs.

carlescufi, 2020-03-03:
> Not a bug in the sample, then, but a bug in the documentation — it would be nice to fix the Getting Started guide so it calls out boards that won't work, or advises you to look at the sample's README (in which case it'd also be nice if the sample mentioned that the MicroBit is not supported.)

I think directing users to the `README.rst` of the sample, or to its rendered version, which in your case is:
https://docs.zephyrproject.org/latest/samples/basic/blinky/README.html

Would you mind sending a Pull Request to the Getting Started Guide for this?

carlescufi, 2020-03-04:
@snej turns out blinky cannot run on the micro:bit at all, because the micro:bit doesn't have a simple LED connected to a GPIO line. It does have however a "display" driver that you can test with this sample:
https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/boards/bbc_microbit/display

I have sent a PR that clarifies the blinky situation: #23262

PR #23262 (2020-03-04): doc: Getting Started: Add a note about building the blinky sample
See commit messages.
CC @snej.

Fixes #23169.
commit cf2d474a (2020-03-04): doc: gsg: Add a note about blinky

Add a note that describes the fact that blinky does not run on all
boards supported by Zephyr, and propose an alternative (Hello World) for
those boards.

Fixes #23169.
closed 2020-03-04 by jhedberg