zephyrproject-rtos/zephyr #23177: Bluetooth: Mesh: Access structure member with a possible NULL pointer

tluio, 2020-02-29:
File: /subsys/bluetooth/host/mesh/cfg_srv.c
Function: friend_set, send_friend_status

On line 2709 of the code, in function friend_set, **model->user_data** is assigned to **cfg**.
On line 2720 of the code, if **cfg** is NULL, jump to label send_status where send_friend_status is called and **model** is passed as the first argument.
On line 2684 of the code, in function send_friend_status, **model->user_data** is assigned to **cfg**. **model** is the first formal parameter, so the local variable cfg is NULL here.
On line 2687 of the code, when calling function net_buf_simple_add_u8, a member is accessed through cfg->frnd, but cfg is NULL here.

`
2705: static void friend_set(struct bt_mesh_model *model,
2706: 		       struct bt_mesh_msg_ctx *ctx,
2707: 		       struct net_buf_simple *buf)
2708: {
2709:	struct bt_mesh_cfg_srv *cfg = model->user_data;
2710: 
2711：	BT_DBG("net_idx 0x%04x app_idx 0x%04x src 0x%04x len %u: %s",
2712：	       ctx->net_idx, ctx->app_idx, ctx->addr, buf->len,
2713：	       bt_hex(buf->data, buf->len));
2714：
2715：	if (buf->data[0] != 0x00 && buf->data[0] != 0x01) {
2716:		BT_WARN("Invalid Friend value 0x%02x", buf->data[0]);
2717:		return;
2718:	}
2719: 
2720: 	if (!cfg) {
2721: 		BT_WARN("No Configuration Server context available");
2722:		goto send_status;
2723: 	}
2724: 
2725： 	BT_DBG("Friend 0x%02x -> 0x%02x", cfg->frnd, buf->data[0]);
2726： 
2727： 	if (cfg->frnd == buf->data[0]) {
2728： 		goto send_status;
2729： 	}
2730： 
2731: 	if (IS_ENABLED(CONFIG_BT_MESH_FRIEND)) {
2732: 		cfg->frnd = buf->data[0];
2733: 
2734: 		if (IS_ENABLED(CONFIG_BT_SETTINGS)) {
2735: 			bt_mesh_store_cfg();
2736: 		}
2737: 
2738: 		if (cfg->frnd == BT_MESH_FRIEND_DISABLED) {
2739: 			bt_mesh_friend_clear_net_idx(BT_MESH_KEY_ANY);
2740: 		}
2741: 	}
2742: 
2743: 	if (cfg->hb_pub.feat & BT_MESH_FEAT_FRIEND) {
2744: 		bt_mesh_heartbeat_send();
2745: 	}
2746: 
2747: send_status:
2748: 	send_friend_status(model, ctx);
2749: }
`
`
2680: static void send_friend_status(struct bt_mesh_model *model,
2681: 			       struct bt_mesh_msg_ctx *ctx)
2682: {
2683: 	BT_MESH_MODEL_BUF_DEFINE(msg, OP_FRIEND_STATUS, 1);
2684: 	struct bt_mesh_cfg_srv *cfg = model->user_data;
2685: 
2686: 	bt_mesh_model_msg_init(&msg, OP_FRIEND_STATUS);
2687: 	net_buf_simple_add_u8(&msg, cfg->frnd);
2688: 
2689： 	if (bt_mesh_model_send(model, ctx, &msg, NULL, NULL)) {
2690： 		BT_ERR("Unable to send Friend Status");
2691： 	}
2692：}
`

aescolar, 2020-03-02:
CC @jhedberg @trond-snekvik

jhedberg, 2020-03-02:
The bug is actually the unnecessary check in `friend_set()` since the `cfg_srv_init()` function ensures that `model->user_data` is always non-NULL. Setting low priority for this one since it's essentially harmless, but we should still strive to eliminate dead code like this.

joerchan, 2020-03-06:
@tluio Not sure why you closed it. We should eliminate the dead code.

tluio, 2020-03-08:
> @tluio Not sure why you closed it. We should eliminate the dead code.

Sorry, I am not familiar with github, so I am not sure when issues should be closed. Thank you for reopening it. I will be more careful next time.

joerchan, 2020-07-16:
Fixed by: https://github.com/zephyrproject-rtos/zephyr/pull/25365

closed 2020-07-16 by joerchan