zephyrproject-rtos/zephyr #23184: mqtt_connect fails with net_sock_tls: TLS handshake error: -2700

md-jamal, 2020-03-01:
Hi Guys,

I tried to connect to MQTT Amazon Web Services IOT Service. Wrote a basic code by following other two cloud samples. Please have a look at the below code

```
#include <logging/log.h>
LOG_MODULE_REGISTER(aws_mqtt, LOG_LEVEL_DBG);

#include <zephyr.h>
#include <net/socket.h>
#include <net/mqtt.h>
#include <net/net_config.h>
#include <net/net_event.h>
#include <net/sntp.h>

#include <sys/printk.h>
#include <string.h>
#include <errno.h>
#include "dhcp.h"
#include <time.h>
#include <inttypes.h>
#include "globalsign.inc"
#if defined(CONFIG_SOCKS)
static struct sockaddr socks5_proxy;
#endif
static struct pollfd fds[1];
static int nfds;

#define CONFIG_SAMPLE_CLOUD_AWS_PASSWORD	"xxxxx"
#define CONFIG_SAMPLE_CLOUD_AWS_USERNAME	"xxxx@gmail.com"

#define APP_MQTT_BUFFER_SIZE    128
#define APP_CA_CERT_TAG 1

static sec_tag_t m_sec_tags[] = {

    APP_CA_CERT_TAG,
};

/* Buffers for MQTT client. */
static u8_t rx_buffer[APP_MQTT_BUFFER_SIZE];
static u8_t tx_buffer[APP_MQTT_BUFFER_SIZE];
static bool connected;

#define MQTT_CLIENTID           "zephyr_publisher"

struct zsock_addrinfo *haddr;
s64_t time_base;
static struct sockaddr_storage broker;
static struct mqtt_client client_ctx;

static void clear_fds(void)
{
	nfds = 0;
}

static void broker_init(void)
{
	struct sockaddr_in *broker4 = (struct sockaddr_in *)&broker;

	broker4->sin_family = AF_INET;
	broker4->sin_port = htons(8883);

	net_ipaddr_copy(&broker4->sin_addr,
			&net_sin(haddr->ai_addr)->sin_addr);

}

void mqtt_evt_handler(struct mqtt_client *const client,
		const struct mqtt_evt *evt)
[... 353 lines cut ...]
CONFIG_NET_CONNECTION_MANAGER_LOG_LEVEL_DBG=n
CONFIG_DNS_RESOLVER_LOG_LEVEL_DBG=n
CONFIG_NET_CONTEXT_LOG_LEVEL_DBG=n
CONFIG_SOC_PART_NUMBER_SAME70Q21B=y
CONFIG_MQTT_LOG_LEVEL_DBG=y
CONFIG_SNTP=y

CONFIG_NET_SOCKETS_POSIX_NAMES=y

CONFIG_MAIN_STACK_SIZE=4096
CONFIG_MQTT_LIB_TLS=y
# Enable TLS credential filenames for secure socket offload
#CONFIG_NET_SOCKETS_OFFLOAD=y

# Enable Mbed TLS configuration
CONFIG_MBEDTLS=y
CONFIG_MBEDTLS_BUILTIN=y
CONFIG_MBEDTLS_ENABLE_HEAP=y
CONFIG_MBEDTLS_HEAP_SIZE=100000
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=10240
CONFIG_MBEDTLS_PEM_CERTIFICATE_FORMAT=y
CONFIG_MBEDTLS_DEBUG=y
CONFIG_MBEDTLS_DEBUG_LEVEL=4

```

rlubos, 2020-03-03:
Please remove the `bug` label unless you have a specific bug report.

The original title suggested that there is a problem with CA certificate used (`0x2700` is `MBEDTLS_ERR_X509_CERT_VERIFY_FAILED`). Did anything change in that matter which caused the title change?

I recommend to enable mbed TLS logs in order to proceed with investigation. Try adding the following:
```
CONFIG_MBEDTLS_DEBUG=y
CONFIG_MBEDTLS_DEBUG_LEVEL=4
CONFIG_NET_SOCKETS_LOG_LEVEL_DBG=y
```

md-jamal, 2020-03-03:
How to enable 'MBEDTLS_X509_CRT_PARSE_C' this particular option 
From subsys/net/lib/sockets/sockets_tls.c, line number 331
```
#if defined(MBEDTLS_X509_CRT_PARSE_C)
		mbedtls_x509_crt_init(&tls->ca_chain);
		mbedtls_x509_crt_init(&tls->own_cert);
		mbedtls_pk_init(&tls->priv_key);
#endif
```

rlubos, 2020-03-03:
It's enabled by default when certificate-based cipher suite is selected:
https://github.com/zephyrproject-rtos/mbedtls/blob/master/configs/config-tls-generic.h#L336

md-jamal, 2020-03-04:
I have enabled CONFIG_MBEDTLS_KEY_EXCHANGE_RSA_PSK_ENABLED=y in prj.conf

Then i have added the following statement in src/main.c

```
	app_dhcpv4_startup();
#if defined(MBEDTLS_X509_CRT_PARSE_C)
	LOG_INF("Should have DHCPv4 lease at this point.");
#endif
```
After flashing the code on to the device, and looking at serial logs, this log is not present, hence my question

md-jamal, 2020-03-05:
Got the zephyr able to communicate with aws, if anyone is interested 

```
#include <logging/log.h>
LOG_MODULE_REGISTER(aws_mqtt, LOG_LEVEL_DBG);

#include <zephyr.h>
#include <net/socket.h>
#include <net/mqtt.h>
#include <net/net_config.h>
#include <net/net_event.h>
#include <net/sntp.h>

#include <sys/printk.h>
#include <string.h>
#include <errno.h>
#include "dhcp.h"
#include <time.h>
#include <inttypes.h>
#include "globalsign.inc"
#if defined(CONFIG_SOCKS)
static struct sockaddr socks5_proxy;
#endif
static struct pollfd fds[1];
static int nfds;

#define ALIVE_TIME      (MSEC_PER_SEC * 60U)
#define APP_MQTT_BUFFER_SIZE    128
#define APP_CA_CERT_TAG 1
#define APP_PRIVATE_SERVER_KEY_TAG 2

static sec_tag_t m_sec_tags[] = {

    APP_CA_CERT_TAG,
    APP_PRIVATE_SERVER_KEY_TAG,
};

/* Buffers for MQTT client. */
static u8_t rx_buffer[APP_MQTT_BUFFER_SIZE];
static u8_t tx_buffer[APP_MQTT_BUFFER_SIZE];
static bool connected;
static u64_t next_alive;
#define MQTT_CLIENTID           "zephyr_publisher"

struct zsock_addrinfo *haddr;
s64_t time_base;
static struct sockaddr_storage broker;
static struct mqtt_client client_ctx;

static void clear_fds(void)
{
	nfds = 0;
}

static void broker_init(void)
{
	struct sockaddr_in *broker4 = (struct sockaddr_in *)&broker;

	broker4->sin_family = AF_INET;
	broker4->sin_port = htons(8883);

	net_ipaddr_copy(&broker4->sin_addr,
			&net_sin(haddr->ai_addr)->sin_addr);

}

void mqtt_evt_handler(struct mqtt_client *const client,
		const struct mqtt_evt *evt)
{
	int err;
[... 303 lines cut ...]
		goto top;
	}

	a = &net_sin(addr->ai_addr)->sin_addr;
	LOG_INF("  Got %s",
			log_strdup(net_addr_ntop(addr->ai_family, a,
					hr_addr, sizeof(hr_addr))));

}

void main(void)
{
	static struct addrinfo hints;
	struct addrinfo *haddr;
	int res;
	int cnt = 0;
        int rc;

	app_dhcpv4_startup();
        rc = tls_init();
        PRINT_RESULT("tls_init", rc);
	mqtt_startup(CONFIG_AWS_HOSTNAME, 8883);	

}
```

jukkar, 2020-03-11:
> Got the zephyr able to communicate with aws, if anyone is interested

Thanks for the example! Would you be able to submit a sample app to `samples/net/clould/` for this so that it would be available also as a part of zephyr?

md-jamal, 2020-03-12:
Do we have any documentation on how to submit?

jukkar, 2020-03-13:
See zephyr documentation for info https://docs.zephyrproject.org/latest/contribute/index.html?highlight=submit

md-jamal, 2020-03-15:
I have created a pull request : https://github.com/zephyrproject-rtos/zephyr/pull/23469
I am doing this first time, please provide feedback and i will correct them

rlubos, 2020-05-05:
The original issue was resolved.

closed 2020-05-05 by rlubos