zephyrproject-rtos/zephyr #23187: nrf_rtc_timer.c  timseout setting mistake.

sbiwa-where, 2020-03-02:
**Describe the bug**
Sleep time is abnormally long.
COUNTER register overruns before setting CC [0] register.
----------------------------------------------------------------
nrf_rtc_timer.c
z_clock_set_timeout()
{
 ...
 (line:145) u32_t cyc, dt, t = counter(); (Ex t=10)
 (line:172) cyc += last_count; (Ex cyc=13)
 (line:200)    if (counter_sub(cyc, t) > 2) { (counter_sub(cyc, t)=3)
  -> interrupt occur
  -> interrput return (Ex count=20)
 (line:201) set_comparator(cyc) CC[0]=13 COUNTER=20
 (line:231) if (zli_fixup && counter_sub(cyc, counter()) <= 0x7fffff) {
  ->  zli_fixup=true counter_sub(13,20)=0xFFFFF9u
RESULT:
 CC[0] = 13
 COUNTER=20
 EVENTS_COMPARE[0]=0
----------------------------------------------------------------
No events for a long time

**To Reproduce**
TARGET BOARD is nrf52833_pca10100. and use NORDIC proprietary ble_controller.

carlescufi, 2020-03-03:
@nordic-krch the nRF Controller uses ZLI, in case this matters.
@sbiwa-where what revision of NCS are you using exactly?

sbiwa-where, 2020-03-03:
I use NCS v1.2.0.

nordic-krch, 2020-03-04:
i will try to port algorithm from `counter_nrfx_rtc` which is lockless so handles ZLI case as well.

nordic-krch, 2020-03-04:
can you try to cherrypick this: https://github.com/zephyrproject-rtos/zephyr/pull/23251/commits/f63f22aebd63b56a7b0ca15dad3dab6c2091135c and check if issue still exists?

PR #23251 (2020-04-22): drivers: timer: nrf_rtc: Refactor alarm setting
User reported a flaw in the current algorithm which fails when Zero
Latency Interrupts (ZLI) are used. Ported algorithm from
counter_nrfx_rtc.c which covers all cases. Algorithm is lockless so
no distinction for ZLI is needed.

Fixes #23187.

Algorithm is ported with modifications from `counter_nrfx_rtc.c` and should cover all corner cases when setting RTC compare register.

I've run tests from `tests/kernel/timer` and `tests/kernel/tickless/tickless_concept`.
closed 2020-04-22 by carlescufi