zephyrproject-rtos/zephyr #23198: rf2xx driver uses mutex in ISR

markus-becker-tridonic-com, 2020-03-02:
**Describe the bug**

The ieee802154_rf2xx driver is using a mutex in ISR context:

https://github.com/zephyrproject-rtos/zephyr/blob/master/drivers/ieee802154/ieee802154_rf2xx.c#L85-L87

With CONFIG_ASSERT=y, the kernel will  report

```
ASSERTION FAIL [!arch_is_in_isr()] @ ZEPHYR_BASE/kernel/include/ksched.h:262
``` 

The MCR driver handles this differently, I think.

FYI: @nandojve

carlescufi, 2020-03-03:
@nandojve will you send a fix for this?

nandojve, 2020-03-03:
@carlescufi I can work but I need some help to understand my options inside a timer callback to change **ctx->trx_state** variable.
```
static void trx_isr_timeout(struct k_timer *timer_id)
{
	struct rf2xx_context *ctx = (struct rf2xx_context *)
				    k_timer_user_data_get(timer_id);

	k_mutex_lock(&ctx->phy_mutex, K_FOREVER);
	ctx->trx_state = RF2XX_TRX_PHY_STATE_IDLE;
	k_mutex_unlock(&ctx->phy_mutex);
}
```
This callback affects transceiver state machine and mistakes around that variable could lead to a deadlock.

Need refactor name from **trx_isr_timeout** to **trx_timeout_callback** too.

carlescufi, 2020-03-03:
@tbursztyka input welcome for this issue, in case you have a suggestion.

nandojve, 2020-03-03:
Can I work on top of https://github.com/zephyrproject-rtos/zephyr/pull/22511?
Is this block release?

PR #23242 (2020-03-09): drivers: ieee802154: Rem trx_state variable
The rx timeout timer callback need update **trx_state** variable and this variable is protected by a **mutex**. Because of that, when compiling the system with **CONFIG_ASSERT=y** the system reports _ASSERTION FAIL [!arch_is_in_isr()] @ ZEPHYR_BASE/kernel/include/ksched.h:262_.

This refactor the driver remove **trx_state** variable dependency and consequently removes **phy_mutex** and **rx timeout timer** to be compliant with kernel rules.

Fixes: #23198

This PR includes #22511
commit 2cfb11e0 (2020-03-09): drivers: ieee802154: rf2xx: Rem trx_state variable

The rx timeout timer callback need update trx_state variable and this
variable is protected by a mutex. Because of that, when compiling the
system with CONFIG_ASSERT=y the system reports 'ASSERTION FAIL
[!arch_is_in_isr()] @ ZEPHYR_BASE/kernel/include/ksched.h:262'.

This refactor the driver remove trx_state variable dependency and
consequently removes phy_mutex and rx timeout timer to be compliant
with kernel rules.

Fixes: #23198
closed 2020-03-09 by jhedberg