zephyrproject-rtos/zephyr #23207: tests/kernel/mem_pool/mem_pool_concept failed on mec15xxevb_assy6853 board.

chen-png, 2020-03-03:
**Describe the bug**
using sanitycheck to run tests/kernel/mem_pool/mem_pool_concept on mec15xxevb_assy6853 board, there are some error log message output, it will cause we can't use the junit2html to convert the sanitycheck.xml to html file.

**To Reproduce**
1. sanitycheck --hardware-map ~/mec1.file --device-testing 
      -T  tests/kernel/mem_pool/mem_pool_concept/ -vv
2. see error
3. console output

DEBUG   - DEVICE: *** Booting Zephyr OS build v2.2.0-rc3  ***
DEBUG   - DEVICE: Running test suite mpool_concept
DEBUG   - DEVICE: ================================================
DEBUG   - DEVICE: starting test - test_mpool_alloc_wait_prio
DEBUG   - DEVICE: PASS - test_mpool_alloc_wait_prio
DEBUG   - DEVICE: ==================================================
DEBUG   - DEVICE: starting test - test_mpool_alloc_size_roundup
DEBUG   - DEVICE: PASS - test_mpool_alloc_size_roundup
DEBUG   - DEVICE: ==================================================
DEBUG   - DEVICE: starting test - test_mpool_alloc_merge_failed_diff_size

**DEBUG   - DEVICE: 6Ly6!
`>6O>6g8K}}}ͥ5)j4хѥѕс}}}ɝ}}}ɕ5r6Ly6!
`>6O>6g8X[kՕɝ}}}ɕ5RjDUkсեѕj}сՍ5RjR%jtԨU**

DEBUG   - Timed out while monitoring serial output on mec15xxevb_assy6853
INFO    - 1/1 mec15xxevb_assy6853       tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept FAILED Timeout (device 112.089s)
ERROR   - see: /home/wentongw/work/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/tests/kernel/mem_pool/mem_pool_concept/kernel.memory_pool.concept/handler.log

INFO    - 0 of 1 tests passed (0.00%), 1 failed, 0 skipped with 0 warnings in 116.61 seconds
INFO    - In total 1 test cases were executed on 1 out of total 229 platforms (0.44%)

    

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 5372e78827

chen-png, 2020-03-09:
this will influence the QA test framework, we can't  use the xml to generate html, so we can't see the detail test result in daily test result and testrail.
 it's critical for QA, so change it's priority to high.

carlescufi, 2020-03-09:
@wentongwu do you think you will have a patch for this soon? This is currently blocking the 2.2 release.

nashif, 2020-03-09:
This is not a high priority bug. I suggest to either exclude the test or manually edit the xml file and remove the garbage.

nashif, 2020-03-09:
my guess is that the issue is related to PM being enabled by default for this board:
CONFIG_SOC_POWER_MANAGEMENT=y

This should never be enabled by default and should be use-case/application specific.

wentongwu, 2020-03-10:
```
diff --git a/soc/arm/microchip_mec/mec1501/power.c b/soc/arm/microchip_mec/mec1501/power.c
index e86dc2f088..9a705926ce 100644
--- a/soc/arm/microchip_mec/mec1501/power.c
+++ b/soc/arm/microchip_mec/mec1501/power.c
@@ -72,6 +72,7 @@ static void z_power_soc_deep_sleep(void)

        soc_deep_sleep_non_wake_dis();

+       k_busy_wait(3000);
        soc_deep_sleep_periph_restore();

 }
```

@chen-png would you like to try this patch, I test it locally it seems working well?

wentongwu, 2020-03-10:
> Did some bisecting and that [ac2b117](https://github.com/zephyrproject-rtos/zephyr/commit/ac2b117acd837b9a08b847ef8a58b870ca96914e) introduced this issue. The test works fine without garbage characters on the commit before this one. This commit re-introduced the deep sleep state residencies.

@dcpleung @nashif without this commit, deep sleep state can't be entered. The bsp code about mec power management is merged on Dec 10, 2019 (see https://github.com/zephyrproject-rtos/zephyr/pull/19423), but removing deep sleep states from residency policy happened on Dec 4, 2019 (see https://github.com/zephyrproject-rtos/zephyr/pull/20950), that means we never test the deep sleep state until commit https://github.com/zephyrproject-rtos/zephyr/commit/ac2b117acd837b9a08b847ef8a58b870ca96914e merged, so we find so many bugs about deep sleep on this board after this commit.

wentongwu, 2020-03-10:
@scottwcpg @dcpleung I don't have the spec about this SOC, but I see some comments in the code like below, and I assume the 48Mhz PLL is related to peripherals, so I add some delay(3ms) after resume to get this PLL locked before uart start to work. 
```
 * Deep Sleep
 * Pros:
 * Lower power dissipation, 48MHz PLL is off
 * Cons:
 * Longer wake latency. CPU start running on ring oscillator
 * between 16 to 25 MHz. Minimum 3ms until PLL reaches lock
 * frequency of 48MHz.
```

chen-png, 2020-03-10:
> @chen-png would you like to try this patch, I test it locally it seems working well?

tested, it worked well for mec board now.

PR #23383 (2020-03-11): soc: mec1501: wait for PLL to lock coming out of deep sleep
The 48MHz PLL on MEC1501 is shut off during deep sleep (i.e. heavy
sleep in datasheet). When coming out of deep sleep, this PLL needs
about 3ms to lock. Most peripherals are using this PLL as clock
source and timing would be off before PLL is locked. Example of
this is seen on serial console where garbage characters are sent
as the UART block is not pushing characters out at the configured
baud rate. This likely affects all other peripherals such as I2C
and eSPI. Luckily, there is a register to indicate whether the PLL
is ready. So sp [... cut]
commit 4700c93f (2020-03-11): soc: mec1501: wait for PLL to lock coming out of deep sleep

The 48MHz PLL on MEC1501 is shut off during deep sleep (i.e. heavy
sleep in datasheet). When coming out of deep sleep, this PLL needs
about 3ms to lock. Most peripherals are using this PLL as clock
source and timing would be off before PLL is locked. Example of
this is seen on serial console where garbage characters are sent
as the UART block is not pushing characters out at the configured
baud rate. This likely affects all other peripherals such as I2C
and eSPI. Luckily, there is a register to indicate whether the PLL
is ready. So s [... cut]
closed 2020-03-11 by nashif