zephyrproject-rtos/zephyr #23208: Can not flash test images into up_squared board.

chen-png, 2020-03-03:
**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto samples/hello_world/
2. west flash
3. See error

**Screenshots or console output**
[ztest@zserver12 zephyr]$ west flash ~/up_squared.sh
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner misc-flasher
flash
**FATAL ERROR: required program flash not found; install it or add its location to PATH**

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 5372e78827a

jhedberg, 2020-03-03:
Looks more like a development environment issue?

dcpleung, 2020-03-03:
The flash script is probably only available on your CI system, and is specific to the DUT. Try finding the script in your CI and see if it works on your dev machine. Note that west flash is not supported on master. The official way is to provision a USB stick with GRUB and the Zephyr binary.

Closing this as this is expected and not a bug.

chen-png, 2020-03-04:
my script is ok to download test image into up_squared via pxe, and it worked well with commit "f8cb3d4d53" in master branch.
for up_squared, misc-flasher was added into master to support sanitycheck running on up_sqaured.

chen-png, 2020-03-04:
@mbolivar  it seems that this issue is related to commit "dc0948424e" in master.
 my added command option "up_sqaured.sh" script in args was replaced by the "flash" command in user_args.

dcpleung, 2020-03-04:
I have verified that `west flash` is broken with dc0948424ef3711e65e7f351dcaa74682ec3eb18.

jhedberg, 2020-03-04:
> I have verified that west flash is broken with dc09484.

@mbolivar FYI

mbolivar, 2020-03-04:
> > I have verified that west flash is broken with [dc09484](https://github.com/zephyrproject-rtos/zephyr/commit/dc0948424ef3711e65e7f351dcaa74682ec3eb18).
> 
> @mbolivar FYI

This issue is specific to misc-flasher, so I've posted a fix to just that file in https://github.com/zephyrproject-rtos/zephyr/pull/23260.

PR #23260 (2020-03-04): runners: misc-flasher: hotfix
The 'command' command line argument for this flasher is now being
overridden by common code, which attaches the west subcommand name to
this.

Let's just hotfix this by renaming the argument in misc-flasher.
We can revisit the boundary between run_common.py arguments and
runners package arguments after the release.

Fixes: #23208
closed 2020-03-04 by jhedberg