zephyrproject-rtos/zephyr #23210: tests/net/iface failed on mimxrt1050_evk board with v1.14 branch.

chen-png, 2020-03-03:
**Describe the bug**
Running tests/net/iface test case on mimxrt1050_evk board with v1.14 branch, it will meet a "BUF FAULT" error.
this is an occasionally error,  similar to #21257.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b=mimxrt1050_evk tests/net/iface
2. west flash
3. See error

**Screenshots or console output**

***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0x2b4f7808
***** Hardware exception *****
Current thread ID = 0x800026ac
Faulting instruction address = 0x60008ce6
Fatal fault in thread 0x800026ac! Aborting.
[***** MPU FAULT *****
  Stacking error (context area might be not valid)
  Data Access Violation
  MMFAR Address: 0x80005810
***** Stack Check Fail! *****
Current thread ID = 0x800025c8
Faulting instruction address = 0x800059a8
Fatal fault in essential thread! Spinning...

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: zephyr-sdk-0.10.3
 - Commit ID: ff682926

nashif, 2020-03-18:
This test is not designed to run on actual platforms, right now it is whitelisted to native_posix qemu_x86 qemu_cortex_m3 only and does unittesting of network interfaces

chen-png, 2020-08-06:
it's still an issue, reopened it

jocelyn-li, 2020-09-28:
@Zhaoningx can this be closed?

Zhaoningx, 2020-09-29:
> @Zhaoningx can this be closed?

Hi @jocelyn-li @chen-png , I saw that the issue has been fixed, so closed this now. :)

closed 2020-09-29 by Zhaoningx