zephyrproject-rtos/zephyr #23214: least-privilege validation framework for kernel object permissions

andrewboie, 2020-03-03:
**Is your enhancement proposal related to a problem? Please describe.**
Programmers are sometimes lazy and may grant threads access to more kernel objects or device drivers than they actually need. It would be great if we could audit the permissions that threads have been given vs. what they actually needed to do their job.

**Describe the solution you'd like**
This sort of accounting will involve extra bookkeeping and overhead, so this should be a non-default debugging configuration option.

All thread access to kernel objects is checked in `z_object_validate()`. Every kernel object metadata struct (`struct _k_object`) has the `perms` bitfield indicating what threads have access to it. Add a parallel bitfield `perms_checked`. In this bitfield, if thread with id N validates access, enable the Nth bit in `perms_checked`.

Then introduce a function `k_thread_perms_audit()` which, when invoked, will iterate over all known kernel objects with `z_object_wordlist_foreach()` and compare the `perms` and `perms_checked` bitfields. If bit N in active in `perms` but not `perms_checked`, that means that at least so far, such permission is not necessary for thread N, and dump appropriate human-comprehensible information about this to the console. Applications will invoke this at a time they deem appropriate.

Additionally add a hook on thread exit to do the same, but just for the exiting thread and not all threads.

**Describe alternatives you've considered**
None, this seems pretty straightforward for any runtime solution.

I don't believe that it's possible to conduct this sort of analysis at build time without some Coverity-level static analysis infrastructure available; doesn't seem practical at this time.

watsonkm, 2023-09-21:
Would love to hop on this if still available!

ceolin, 2023-09-22:
@foxmoder that is still in the backlog. Fell free to jump on it.