zephyrproject-rtos/zephyr #23215: fujitsu FRAM read error on stm32_olimexino

boonpin1975, 2020-03-03:
on zephyr 2.1.0 using fram model https://learn.adafruit.com/adafruit-spi-fram-breakout
on olimexino stm32 , 
�*** Booting Zephyr OS build zephyr-v2.1.0-1662-ga28f8fbef542  ***
fujitsu FRAM example application
Error during ID read
Could not verify FRAM ID

is this the model supported ? thanks !

ydamigos, 2020-03-03:
Yes, this is the supported model. `spi_fujitsu_fram` sample supports MB85RS64V FRAM and it the one used on adafruit SPI FRAM breakout.
Did you connect olimexino_stm32 to adafruit SPI FRAM breakout board correctly? By default SPI_1 is configured on UEXT connector on olimexino_stm32.

boonpin1975, 2020-03-03:
thanks for you clarification , I am connecting the pin to D10 to D13 on
con4 digital , I will try on the UEXT .

3

D10 (#SS1)

PA4 / SPI1_NSS / USART2_CK / ADC12_IN4

4

D11 (MOSI1)

PA7 / SPI1_MOSI / ADC12_IN7 / TIM3_CH2 / TIM1_CH1N

5

D12 (MISO1)

PA6 / SPI1_MISO / ADC12_IN6 / TIM3_CH1 / TIM1_BKIN

6

D13 (SCK1 / LED1)

PA5 / SPI1_SCK / ADC12_IN5

boonpin1975, 2020-03-04:
I connect to the uext the spi in olimexino , but the fram still can not detected 

�*** Booting Zephyr OS build zephyr-v2.1.0-1662-ga28f8fbef542  ***
fujitsu FRAM example application
Error during ID read
Could not verify FRAM ID

ydamigos, 2020-03-04:
Do you have a logic analyzer to check the signals?

boonpin1975, 2020-03-04:
I just use a loopback ( miso connect to mosi ) to check if it's working 
[image]

ydamigos, 2020-03-05:
How do you configure CS signal? Does the application drive it or the hardware (CONFIG_SPI_STM32_USE_HW_SS)?

boonpin1975, 2020-03-05:
I change to the stm32 blue pill , but it still not accessible to this module 

�*** Booting Zephyr OS build v2.2.0-rc3-43-gf7564323f592  ***                   
fujitsu FRAM example application                                                
Error during ID read                                                            
Could not verify FRAM ID

boonpin1975, 2020-03-05:
I am using the out of box prj.conf from
https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/drivers/spi_fujitsu_fram/prj.conf

CONFIG_STDOUT_CONSOLE=y
CONFIG_PRINTK=y

CONFIG_SPI=y
CONFIG_GPIO=y

boonpin1975, 2020-03-06:
I added software cs , connected to PA11 in blue pill but it did not work either , to do a comparison , this fram is working when I connected to the arduino . thanks 

// PA11
	spi_cs.gpio_dev = device_get_binding("GPIOA");
	spi_cs.gpio_pin = 11;
	spi_cs.delay = 0;

	spi_cfg.cs = &spi_cs;

erwango, 2020-05-11:
@boonpin1975 do you still face this issue ?

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi