zephyrproject-rtos/zephyr #23222: Bluetooth: host: Unable to pair when privacy feature is disabled by application

joerchan, 2020-03-03:
**Describe the bug**
The local on-air address will not be properly initialized when CONFIG_BT_PRIVACY is enabled but the privacy of the device has been disabled. This can happen in two ways:
 - The controller was not instructed to use RPAs because BT_LE_ADV_OPT_USE_IDENTITY was used.
 - An all zeroes local IRK has been set for the identity being used.

This happens because the local RPA will be set to all zeroes by the controller in this case:

This is only valid when the Own_Address_Type (...) is set to 0x02 or
0x03, and the Controller generated a resolvable private address for the
local device using a non-zero local IRK.

**To Reproduce**
Steps to reproduce the behavior:
Using the shell:
Shell 1#:
```
bt init
bt advertise on identity
```
Shell 2#
```
bt init
bt connect <identity of shell #1>
bt security 2
```

**Expected behavior**
When controller reports an all-zeroes local RPA in the connection complete event the host shall
assume that the identity address was used instead.

**Impact**
Unable to perform pairing procedure since the address is input to the SMP pairing functions.

**Screenshots or console output**
Shell 1#
```
uart:~$ bt advertise on identity
Advertising started
Connected: 58:44:1c:d4:ca:a7 (random)
Security failed: 58:44:1c:d4:ca:a7 (random) level 1 reason 1
LE conn param updated: int 0x0028 lat 0 to 42
uart:~$ bt info
Type: LE, Role: slave, Id: 0
Remote address: 58:44:1c:d4:ca:a7 (random) (resolvable)
Local address: f1:c1:d9:aa:49:e2 (random) (identity)
Remote on-air address: 58:44:1c:d4:ca:a7 (random) (resolvable)
Local on-air address: 00:00:00:00:00:00 (random) (non-resolvable)
Interval: 0x0028 (50 ms)
Latency: 0x0000 (0 ms)
Supervision timeout: 0x002a (420 ms)
```
Shell 2#
```
uart:~$ bt connect f1:c1:d9:aa:49:e2 (random)
Connection pending
Connected: f1:c1:d9:aa:49:e2 (random)
Remote LMP version unknown (0x0b) subversion 0xfffe manufacturer 0x0059
LE Features: 0x0000000000004925 
uart:~$ bt security 2
Security failed: f1:c1:d9:aa:49:e2 (random) level 1 reason 1
[00:00:50.567,749] <err> bt_smp: reason 0xb
```

**Additional context**
Add any other context about the problem here.

jhedberg, 2020-03-03:
Reduced the priority since @joerchan reported that this hasn't been reproduced with the upstream tree (the upstream controller, in particular)

joerchan, 2020-03-03:
Reproduced with upstream, but the medium priority was wrong to begin with. This problem has been there since privacy feature was added and can easily be prevented by simply not using the advertise Identity option.

PR #23228 (2020-03-10): Bluetooth: host: Fix local RPA not valid in connection complete
Fix connection complete event handling when the local RPA is not valid.
This can happen when the controller was not instructed to use an RPA
address, or the local IRK was set to an all zero IRK.

Fixes: #23222
closed 2020-03-10 by jhedberg