zephyrproject-rtos/zephyr #23225: Function for updating channel map from environment

BOZHENG001, 2020-03-03:
**Is your feature request related to a problem? Please describe.**
I am now working on BLE development on Zephyr (1.14 and 2.0), I would like to update channel map from environment so that I can add some specific features to my application, but I found no function in Zephyr supporting this. There are only some functions for setting and reading channel map.

**Describe the solution you'd like**
Since BLE channel hopping is based on channel map, and theoretically the channel map should change with surroundings by time. It would be useful if there is a function in Zephyr to update the channel map from environment.

**Additional context**
Previously I asked for a function for reading channel map being used now, because I supposed Zephyr RTOS has been updating channel map from environment periodically or randomly. But actually Zephyr does not really have anything for that yet.

aescolar, 2020-03-03:
CC @jhedberg @joerchan @cvinayak

BOZHENG001, 2020-03-09:
I am not sure if devices from Nordic support environment sensing. I have this feature request, because I suppose that nRF52840 DK has ability to sense environment. If this feature request is impossible on Nordic devices, then please tell me and I will close it.

joerchan, 2020-03-09:
@BOZHENG001 Detecting the channel performance is possible. But there is no standard way over HCI to communicate this information from the controller to the host. So first thing to do would be to come up with a proposal on how to address this issue.

It's the central (or master) role that sets the channel map of the connection. Are you making a central role application?

BOZHENG001, 2020-03-10:
@joerchan Yes, I am making a master application. In BLE stack, only master devices are allowed to update channel map, so I am doing all the procedures in master devices.

carlescufi, 2020-04-07:
Clarification: This enhancement only covers setting the channel map from the Host, not detecting the noise levels in the channels.

carlescufi, 2020-04-08:
@joerchan reminds me that we have `bt_le_set_chan_map()` in the tree today. The ability to update the channel map from noise in the environment is currently not in scope. @BOZHENG001 if you are interested in moving this forward please send a proposal, thanks!

closed 2020-04-08 by carlescufi

BOZHENG001, 2021-03-25:
@carlescufi  Hi carlescufi, it has been a long time since this issue was closed. I am reopening it because I would like to recommend a proposal/paper providing a potential solution for this issue. It introduces both an interference detection method and an improved channel selection algorithm (please see: [https://www.mdpi.com/1424-8220/21/7/2257](url)).

joerchan, 2021-03-25:
@BOZHENG001 You didn't reopen it though. What's your plan forward?

BOZHENG001, 2021-03-25:
@joerchan Indeed I did not, just wanted to share with you about this proposal, since as far as I know, even now there is no standard methodology in the BLE specification. So maybe this can be a future solution for BLE.