zephyrproject-rtos/zephyr #23231: RISCV Machine Timer consistently interrupts long running system after soft reset

jaronkelleher, 2020-03-03:
On a long running riscv platform, the microprocessor may get a soft reset, while the rest of the platform is not reset. In this scenario, the riscv machine timer driver does not function properly. After the first tick interrupt, the timer will constantly interrupt the microprocessor due to invalid mtimecmp register values.

Steps to reproduce the behavior:
1. Run the system long enough for the mtime register to be greater than 0x2_0000_0000
2. soft reset the microprocessor (setting its PC back to __start)
3. Allow the system to boot
4. Witness as the system is constantly interrupted by the timer.

**Expected behavior**
I expect the microprocessor to start normally

**Impact**
Showstopper without the fix

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK
 - Commit SHA: 2709b14bcb368dcc5a0e61783bcff7a04b0488c6

jaronkelleher, 2020-03-03:
Associated with pull request #23201

jhedberg, 2020-03-03:
@jaronkelleher if #23201 will fix this, could you add the appropriate "Fixes #23231" to its description so that GitHub will auto-close this one once it's merged.

PR #23263 (2020-03-04): riscv machine timer: Timer functions on long running platforms
When debugging on a long running platform, the MCU may get reset by
the debugger with an ndmreset toggle. Since there is no requirement
that this resets anything in particular on the platform, the CLINT
registers may not get reset. When this occurs with an mtime register
value that is larger than 32 bits the riscv machine timer will
continuously interrupt the system when the mtime register exceeds 32
bits in value. This is because the last_count value is used to update
the mtimecmp register, and its value is initialized to zero. It [... cut]
closed 2020-03-04 by jhedberg