zephyrproject-rtos/zephyr #23244: kernel.scheduler fails on frdmkw41z

hakehuang, 2020-03-04:
**Describe the bug**
if enable timeslice this case will fails on frdmkw41z. this issues exist on all v2.2.0-rc1 to rc3.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=frdmkw41z ..
3. make flash
4. See error

**Expected behavior**
test pass

**Impact**
kernel threads

**Screenshots or console output**
```
*** Booting Zephyr OS version 2.2.0-rc3  ***
Running test suite threads_scheduling
===================================================================
starting test - test_bad_priorities
PASS - test_bad_priorities
===================================================================
starting test - test_priority_cooperative
PASS - test_priority_cooperative
===================================================================
starting test - test_priority_preemptible
PASS - test_priority_preemptible
===================================================================
starting test - test_yield_cooperative
PASS - test_yield_cooperative
===================================================================
starting test - test_sleep_cooperative
PASS - test_sleep_cooperative
===================================================================
starting test - test_sleep_wakeup_preemptible
PASS - test_sleep_wakeup_preemptible
===================================================================
starting test - test_pending_thread_wakeup
PASS - test_pending_thread_wakeup
===================================================================
starting test - test_time_slicing_preemptible
PASS - test_time_slicing_preemptible
===================================================================
starting test - test_time_slicing_disable_preemptible
PASS - test_time_slicing_disable_preemptible
===================================================================
starting test - test_lock_preemptible
PASS - test_lock_preemptible
===================================================================
starting test - test_unlock_preemptible
PASS - test_unlock_preemptible
===================================================================
starting test - test_unlock_nested_sched_lock
PASS - test_unlock_nested_sched_lock
===================================================================
starting test - test_sched_is_preempt_thread
PASS - test_sched_is_preempt_thread
===================================================================
starting test - test_slice_reset
PASS - test_slice_reset
===================================================================
starting test - test_slice_scheduling
ABCDEFGHI
ABC
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

D
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

E
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

F
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

G
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

H
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

I
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

A
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

B
    Assertion failed at ZEPHYR_BASE/tests/kernel/sched/schedule_api/src/test_slice_scheduling.c:61: thread_tslice: ((tdelta >= expected_slice_min) && (tdelta <= expected_slice_max) && (idx == thread_idx)) is false

FAIL - test_slice_scheduling
===================================================================
starting test - test_priority_scheduling
PASS - test_priority_scheduling
===================================================================
starting test - test_wakeup_expired_timer_thread
PASS - test_wakeup_expired_timer_thread
===================================================================
starting test - test_user_k_wakeup
PASS - test_user_k_wakeup
===================================================================
starting test - test_user_k_is_preempt
PASS - test_user_k_is_preempt
===================================================================
Test suite threads_scheduling failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - v2.2.0-rc3

hakehuang, 2020-03-04:
@MaureenHelm

MaureenHelm, 2020-03-04:
Duplicate of #22745

closed 2020-03-04 by MaureenHelm