zephyrproject-rtos/zephyr #23248: Add secure version of strcpy

yashi, 2020-03-04:
We luck `strlcpy()` or `strscpy()` and its friend.  As a secure RTOS, we should have at least one of those string copy function.  Rationale behind the new APIs are that neither standard `strcpy()` nor `strncpy()` does not guarantee the terminal NUL character.

 - `strlcpy()` is more widely used variant from OpenBSD[1].  It returns the length of the source string.
- `strscpy()` is Linux kernel version of it[2].  It doesn't touch the source string past the specified size and returns negative when error.

There are some discussion on both OpenBSD side[3] and Linux kernel side[4], but both functions make sure that destination string is NUL terminated.

[1]: https://www.usenix.org/legacy/event/usenix99/full_papers/millert/millert.pdf
[2]: https://www.kernel.org/doc/htmldocs/kernel-api/API-strscpy.html
[3]: https://marc.info/?l=openbsd-tech&m=154859632818332&w=2
[4]: https://lkml.org/lkml/2015/4/28/570

vaussard, 2020-03-17:
You have an implementation of `strlcpy()` in Zephyr SDK:
```
arm-zephyr-eabi/include/string.h:116:size_t     strlcpy (char *, const char *, size_t);
```
but it is hidden by default. You have to `#define _GNU_SOURCE` before including `string.h`.

nashif, 2020-08-04:
this should not be part of zephyr, you can have this as part of your application/environment.

stephanosio, 2021-05-27:
TODO:
1. Consider adding a config to enable the relevant functions in newlib.
2. Consider adding tests for these as part of the newlib tests.

yashi, 2021-05-28:
A few questions:
 - Is it just for newlib? I don't really follow #3102 but does it still hold?
 - What has changed since https://github.com/zephyrproject-rtos/zephyr/issues/23248#issuecomment-668721295?

yashi, 2024-02-22:
I'm closing this since picolibc is the default c library for Zephyr for a while now. And it at least provides `strlcpy()`.

closed 2024-02-22 by yashi