zephyrproject-rtos/zephyr #23268: Unnecessary privileged stacks with CONFIG_USERSPACE=y

bbolen, 2020-03-04:
**Describe the bug**
When building with userspace enabled, gen_priv_stacks.py generates privileged stacks for system stacks that do not need it.  For example, on Cortex-M platforms, there is no need for a privileged shadow stack of _interrupt_stack, z_main_stack, or z_idle_stack.

**To Reproduce**
Using recent code as of g219d9fc0829
west build -p auto --board=stm32f411e_disco samples/userspace/prod_consumer

~/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-nm build/zephyr/zephyr.elf |grep -e "priv_stack_[0-9]\+"
20004ea0 b priv_stack_20001800
20004aa0 b priv_stack_20002000
200046a0 b priv_stack_20002800
200042a0 b priv_stack_20003000
20003ea0 b priv_stack_20003400

~/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-nm build/zephyr/zephyr.elf |grep -e _interrupt_stack -e z_main_stack -e z_idle_stack
20002800 B _interrupt_stack
20003400 B z_idle_stack
20003000 B z_main_stack

**Impact**
This does not really cause any functional issues but does waste memory which can be scarce on certain platforms.  I can postprocess the elf and remove the unneeded stacks to get around this for now.

I assume that the proper fix would involve fixing up gen_priv_stacks.py to somehow bypass the system related stacks, but I am not sure of a clean way to do that.  That seems like it would need a blacklist of the system stacks.  Another option I was thinking about would create a new macro and type for the system stacks, but that would require touching a lot of core code and does not seem like a good option either.

I'm willing to go attack this problem if I can get a better course of action that the above ideas.

Thanks.

ioannisg, 2020-03-05:
Duplicate of #16210, and will be solved by #13637

ioannisg, 2020-03-05:
@bbolen maybe close this one (you could add this report info in #16210)

bbolen, 2020-03-05:
Sorry for the duplicate.  Thanks for the response.

closed 2020-03-05 by bbolen