zephyrproject-rtos/zephyr #23274: power: subsystem: Application hangs when logging is enabled after entering deep sleep

albertofloyd, 2020-03-05:
When logging is enabled and used in app that sleeps periodically for enough time to enter deep sleep (Sleep time higher than minimum residency)

After entering deep sleep the application thread never execute again which can be perceived as application hang.

Note: from kernel debug traces, deep sleep is exited but immediately re-entered, giving no opportunity for other threads to run.

Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=mec15xxevb_assy6853
3. make
"Wake from light sleep" will be seen, 
"Wake from deep sleep" won't be seen

**Expected behavior**
Board test sleeps for different duration in a loop.
1st case app sleeps for 500 ms more than minimum light sleep residency.
2nd case app sleeps for 500 ms more than minimum deep sleep residency.

Note: Issue is not reproducible if logging is either completely disabled or not used in between k_sleep() during the test.

**Impact**
High cannot enable/use logging simultaneously with deep sleep. 

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

wentongwu, 2020-03-17:
```
#ifdef CONFIG_KERNEL_DEBUG
#define K_DEBUG(fmt, ...) printk("[%s]  " fmt, __func__, ##__VA_ARGS__)
#else
#define K_DEBUG(fmt, ...)
#endif
```
if enabled CONFIG_KERNEL_DEBUG, it will use printk to output debug message. But for your test case, you also enable logging system with CONFIG_LOG=y, I think we should same mechanism to do the output, so could you please try below patch with latest commit, I test it locally, there is no hang like the description. 

```
diff --git a/samples/boards/mec15xxevb_assy6853/power_management/prj.conf b/samples/boards/mec15xxevb_assy6853/power_management/prj.conf
index 328bcabbec..fc2b07a5e6 100644
--- a/samples/boards/mec15xxevb_assy6853/power_management/prj.conf
+++ b/samples/boards/mec15xxevb_assy6853/power_management/prj.conf
@@ -6,6 +6,7 @@ CONFIG_LOG_PROCESS_THREAD_SLEEP_MS=2100
 
 # Enable kernel debug
 CONFIG_KERNEL_DEBUG=y
+CONFIG_LOG_PRINTK=y
 CONFIG_THREAD_NAME=y
 # Enable PM debug in MEC1501 this will give indication in MCHP_060
 CONFIG_SYS_PM_DEBUG=y
```

wentongwu, 2020-03-17:
@albertofloyd I will submit a PR.

albertofloyd, 2020-03-20:
@wentongwu  are you saying that the problem is due to wrong combination of debug options in KConfig?  I will try the PR

wentongwu, 2020-03-21:
> @wentongwu are you saying that the problem is due to wrong combination of debug options in KConfig? I will try the PR

@albertofloyd yes, please try it, thanks.

PR #24254 (2020-04-10): soc: mec1501: modifies interrupt restoration after deep sleep
z_power_soc_deep_sleep() is called with interrupt locked already
so restoring BASEPRI is pointless here, as it would only allow
exceptions afterwards. The situation is complicated by the fact
that kernel/idle.c:idle() only locks interrupt without unlocking
which means the BASEBRI at entry of z_power_soc_deep_sleep() is
already set to allow exceptions only but not lower priority
interrupts like timer. So when, e.g. timer, interrupt fires,
the SoC would come out of deep sleep but the waking interrupts
are never delivered since they ar [... cut]
commit b9359035 (2020-04-10): soc: mec1501: modifies interrupt restoration after deep sleep

z_power_soc_deep_sleep() is called with interrupt locked already
so restoring BASEPRI is pointless here, as it would only allow
exceptions afterwards. The situation is complicated by the fact
that kernel/idle.c:idle() only locks interrupt without unlocking
which means the BASEBRI at entry of z_power_soc_deep_sleep() is
already set to allow exceptions only but not lower priority
interrupts like timer. So when, e.g. timer, interrupt fires,
the SoC would come out of deep sleep but the waking interrupts
are never delivered since they a [... cut]
closed 2020-04-10 by nashif