zephyrproject-rtos/zephyr #23284: driver: ethernet: Add support for a second ethernet controller in the MCUX Ethernet driver

armandciejak, 2020-03-05:
**Is your enhancement proposal related to a problem? Please describe.**
MCUs from the i.MX RT1060 family are equipped with 2 Ethernet controllers but the MCUX driver does support only the first one (ENET base register is hard-coded).

**Describe the solution you'd like**
Modify the driver so that it can support multiple controllers.

**Additional context**
I have a proof of concept barely working, that I can share in order to discuss how to properly implement the changes (I've modified DTS and the driver itself).

jukkar, 2020-03-05:
>  I can share in order to discuss how to properly implement the changes

That is a good idea so please share your changes.

armandciejak, 2020-03-09:
@jukkar I'm cleaning up my proof of concept so that it can be merged. It is not finished yet but you can already have a look here: [eth_mcux_enet2](https://github.com/riedonetworks/zephyr/tree/eth_mcux_enet2)

agansari, 2020-03-09:
@armandciejak good to see you are making this driver multi-instance!

armandciejak, 2020-03-09:
@agansari I'm changing quite a few things, shall I already open a pull request even if I'm not finished with the changes? I would like to get comments ASAP.

agansari, 2020-03-09:
@armandciejak sure, think you can tag it as WIP or DNM until it's close to a merge state

jukkar, 2020-03-09:
> @armandciejak sure, think you can tag it as WIP or DNM until it's close to a merge state

One option is also to create the PR as a draft.

PR #23343 (2020-03-20): Add support for a second Ethernet controller in the MCUX driver
Closes #23284.

This PR allows using the second Ethernet controller of the i.MXRT1060 family.
closed 2020-03-20 by carlescufi