zephyrproject-rtos/zephyr #23292: Bluetooth: Mesh and SMP DHKey coexistence

joerchan, 2020-03-05:
**Describe the bug**
Bluetooth SMP protocol and Bluetooth Mesh provisioner both use the bt_pub_key_gen API.
This API uses the HCI interface to generate a single public/private key pair. The public key is given to the remote, and the private key is used when received the remote public key to generate the DHKey.

While SMP only generates a single public key during init Mesh will generate a new one for each provisioning.

These two use-cases of the DHKey functionality in HCI cannot coexist because the HCI interface only support a single-instance public/private key-pair.
So we cannot support generating DHKeys using two different public keys

**To Reproduce**
Steps to reproduce the behavior:
Enable both Bluetooth Mesh and SMP protocols.
Perform provisioning.
Pairing cannot be performed anymore since private key used in DHKey is not valid for the public key SMP sent to the remote.

**Expected behavior**
If we are to claim both protocols working simultaneously the DHkey calculations have to be independent.

**Impact**
Bluetooth Mesh and SMP cannot both be enabled.

**Suggested fix**
If we wish to support Mesh simultaneous with regular Bluetooth connections we need to solve the  structural dependency of the HCI commands. This might mean we cannot  rely on the HCI interface and instead move this functionality to the host and change the existing internal ECC API.

**Environment:**
 - Commit: 1.14 or 2.1.0 (or any other commit supporting both protocols).

joerchan, 2020-03-05:
@jhedberg @trond-snekvik Do you foresee this as a possible use-case?

jhedberg, 2020-03-09:
@joerchan at least I haven't come across any device where you'd want both enabled, however I'm not opposed to making improvements to simultaneous access to ECC.

jhedberg, 2020-03-09:
Actually, I'll recategorise this as a feature request, since the current limitations are known

trond-snekvik, 2020-03-09:
We haven't had any feature request for this before in Nordic, but it's not an unthinkable scenario. Perhaps the kconfig entry for SMP should have a `depends on !BT_MESH` to clearly communicate that this isn't supported for now?

joerchan, 2020-03-13:
@trond-snekvik @jhedberg 
I rewrote the issue to be an enhancement instead of a bug report as well. And created #23440
I have heard about one use-case for this, but haven't had a request for zephyr yet.

alxelax, 2023-03-28:
I'm pretty sure in that mesh should have its own dh pair instance. I'll try to refactor mesh crypto engine to get rid of host dependency in dh keys aspect.

alxelax, 2023-03-28:
Even more mesh already has its own function to generate DH public\private keys for OOB public key transferring. I'll reuse it.