zephyrproject-rtos/zephyr #23294: [Coverity CID :208677] Unchecked return value in drivers/sensor/lis3mdl/lis3mdl_trigger.c

zephyrbot, 2020-03-05:
Static code scan issues found in file:

https://github.com/zephyrproject-rtos/zephyr/tree/23ace0e55681c02a81fb6b891f832aa356c85aab/drivers/sensor/lis3mdl/lis3mdl_trigger.c#L28

Category: Error handling issues
Function: `lis3mdl_trigger_set`
Component: Drivers
CID: [208677](https://scan9.coverity.com/reports.htm#v29726/p12996/mergedDefectId=208677)

Details:

```
22         struct lis3mdl_data *drv_data = dev->driver_data;
23         s16_t buf[3];
24    
25         __ASSERT_NO_MSG(trig->type == SENSOR_TRIG_DATA_READY);
26    
27         /* dummy read: re-trigger interrupt */
>>>     CID 208677:  Error handling issues  (CHECKED_RETURN)
>>>     Calling "i2c_burst_read" without checking return value (as is done elsewhere 47 out of 57 times).
28         i2c_burst_read(drv_data->i2c, DT_INST_0_ST_LIS3MDL_MAGN_BASE_ADDRESS,
29                    LIS3MDL_REG_SAMPLE_START, (u8_t *)buf, 6);
30    
31         gpio_pin_interrupt_configure(drv_data->gpio,
32                 DT_INST_0_ST_LIS3MDL_MAGN_IRQ_GPIOS_PIN,
33                 GPIO_INT_DISABLE);
```

Please fix or provide comments in coverity using the link:

https://scan9.coverity.com/reports.htm#v32951/p12996.

Note: This issue was created automatically. Priority was set based on classification
of the file affected and the impact field in coverity. Assignees were set using the CODEOWNERS file.

ceolin, 2020-04-16:
@avisconti  It seems that is irrelevant the return of `i2c_burst_read`, so it should be fine just cast it void to explicitly ignore. What is looking wrong to me is the following part:
```
       drv_data->data_ready_handler = handler;
        if (handler == NULL) {
                return 0;
        }
```
Is that valid ? Shouldn't we return `-EINVAL` here ?

avisconti, 2020-04-17:
> @avisconti It seems that is irrelevant the return of `i2c_burst_read`, so it should be fine just cast it void to explicitly ignore. What is looking wrong to me is the following part:
>
> ``` [...]

@ceolin 
To be honest I think the caller is allowed to set an interrupt w/o the need for a handler to be called. Maybe the issue here is that we should proceed in setting the interrupt, but when an irq goes off we need to test if handler != NULL in the isr (which is actually tested in lis3mdl_thread_cb).

The bmi160 works in this way. What do you think?

closed 2020-04-20 by carlescufi

ceolin, 2020-04-20:
@avisconti thanks for your help !

avisconti, 2020-04-20:
> @avisconti thanks for your help !

You are welcome! Thanks to you!

PR #24472 (2020-04-20): sensor: lis3mdl: Check possible error in trigger_set
Check the return of i2c_burst_read return and in case of error return
early and propagate the error.

Fixes: #23294
commit 1572106e (2020-04-20): sensor: lis3mdl: Check possible error in trigger_set

Check the return of i2c_burst_read return and in case of error return
early and propagate the error.

Fixes: #23294