zephyrproject-rtos/zephyr #23299: Some bugs or dead codes cased by possible NULL pointers

tluio, 2020-03-05:
I have found some bugs cased by possible NULL pointers. Otherwise, if these pointers cannot be NULL, there will be some dead codes. But I think some of them may be NULL in some special cases.

1.
File: subsys/fb/cfb.c
Function: cfb_print
Code:
108	        const struct char_framebuffer *fb = &char_fb;
109          const struct cfb_font *fptr = &(fb->fonts[fb->font_idx]);
110
111	        if (!fb->fonts || !fb->buf) {
112		         return -1;
113	         }
Description: **fb->fronts** on line 111 cannot be NULL, or bug will occur on line 109 because a member of NULL pointer is accessed through **fb->fonts[fb->font_idx]**. 

2.
File: subsys/net/lib/coap/coap.c
Function: coap_packet_parse
Code:
547          if (options) {
548		          memset(options, 0, opt_num * sizeof(struct coap_option));
549	        }
                ...... ......
579	        while (1) {
580		           struct coap_option *option;
581
582		           option = num < opt_num ? &options[num++] : NULL;
Description: **options** on line 547 cannot be NULL, or bug will occur on line 582 because a member of NULL pointer is accessed through **options[num++]** 

3.
File: subsys/net/l2/ieee802154/ieee802154_mgmt.c
Function: ieee802154_scan
Code:
116   	ctx->scan_ctx = scan;
117  	ret = 0;
118
119  	ieee802154_filter_pan_id(iface, IEEE802154_BROADCAST_PAN_ID);
120
121  	if (ieee802154_start(iface)) {
122  		NET_DBG("Could not start device");
123  		ret = -EIO;
124  
125  		goto out;
126  	}
127 
128  	/* ToDo: For now, we assume we are on 2.4Ghz
129  	 * (device will have to export capabilities) */
130  	for (channel = 11U; channel <= 26U; channel++) {
131  		if (IEEE802154_IS_CHAN_UNSCANNED(scan->channel_set, channel)) {
132  			continue;
133  		}
134
135  		scan->channel = channel;
136  		NET_DBG("Scanning channel %u", channel);
137  		ieee802154_set_channel(iface, channel);
138 
139  		/* Active scan sends a beacon request */
140  		if (mgmt_request == NET_REQUEST_IEEE802154_ACTIVE_SCAN) {
141  			net_pkt_ref(pkt);
142  			net_pkt_frag_ref(pkt->buffer);
143
144  			ret = ieee802154_radio_send(iface, pkt, pkt->buffer);
145  			if (ret) {
146  				NET_DBG("Could not send Beacon Request (%d)",
147  					ret);
148  				net_pkt_unref(pkt);
149  
150  				break;
151  			}
152  		}
153 
154  		/* Context aware sleep */
155  		k_sleep(scan->duration);
156  
157  		if (!ctx->scan_ctx) {
158  			NET_DBG("Scan request cancelled");
159  			ret = -ECANCELED;
160 
161  			break;
162  		}
163  	}
Description: **ctx->scan_ctx** cannot be NULL. On line 116, **scan** is assigned to **ctx->scan_ctx**, so if **ctx->scan_ctx** is NULL, **scan** is NULL, too. But On line 155, a member is accesed through **scan->duration** when **k_sleep** is called, so **scan** cannot be NULL, or bug will occur.

4.
File: subsys/bluetooth/mesh/cfg_srv.c
Function: bt_mesh_friend_get
Code:
3419	BT_DBG("conf %p conf->frnd 0x%02x", conf, conf->frnd);
3420
3421	if (conf) {
3422		return conf->frnd;
3423	}
3424
3425	return BT_MESH_FRIEND_NOT_SUPPORTED;
Description: **conf** on line 3421 cannot be NULL, or bug will occur on line 3419 because a member of NULL pointer is accessed through **conf->frnd**.

jukkar, 2020-03-05:
> 2. Description: **options** on line 547 cannot be NULL, or bug will occur on line 582 because a member of NULL pointer is accessed through **options[num++]**

Options can be NULL, in which case the opt_len should have been set to 0. As the num variable in line 577 is set to 0, then option would evaluate NULL and we do not access options[]. Only problem is that if the caller sets options to NULL and opt_len > 0 then we have a null pointer access. We could check that in this function.

joerchan, 2020-03-05:
> File: subsys/bluetooth/mesh/cfg_srv.c
Function: bt_mesh_friend_get
Code:

This is really a check if the cfg_srv has been initialized. Because cfg_srv cannot be initialized without a valid conf. So just need to update the BT_DBG statement here.
https://github.com/zephyrproject-rtos/zephyr/pull/23301

joerchan, 2020-03-05:
@tluio  Is this a static code analysis tool? Or did you just look around? I'm curious which tool you used,  Shouldn't coverity find these?

tluio, 2020-03-06:
@joerchan I used a static code analysis tool written by myself. But there are still some bugs in it to be fixed.

nashif, 2020-03-06:
> @joerchan I used a static code analysis tool written by myself. But there are still some bugs in it to be fixed.

@tluio  care to share some more information about the tool?

tluio, 2020-03-06:
> > @joerchan I used a static code analysis tool written by myself. But there are still some bugs in it to be fixed.
> 
> @tluio care to share some more information about the tool?

I am so sorry, but I cannot share too much information since it's part of an unpublished work.

PR #23508 (2020-03-17): Fix possible NULL pointers de-references spotted in #23299
Fix possible NULL pointers de-references spotted in #23299

This pr address issues 1 and 3. Issue for was already fixed in c97e1aafb124e1906c951721eb2e9bb2b8813a8b and issue 2 would be a problem only due bad API usage.

Fixes #23299
commit 6fec8ea8 (2020-03-17): net: ieee802154: Fix possible null pointer de-reference

ieee802154_scan() checks if ctx->scan_ctx (scan) is NULL what implies
that this can be true, but de-reference this variable before this
check what may cause a problem.

Fixes #23299 [3]
closed 2020-03-17 by MaureenHelm