zephyrproject-rtos/zephyr #23309: Sanitycheck generated incorrect acrn.xml on acrn platform

chen-png, 2020-03-06:
**Describe the bug**
Running sanitycheck on acrn platform, the generated two files acrn.xml and sanitycheck.xml are not same in sanity-out directory, the test results in these two files are different.

the result in sanitycheck.xml should be right, it's the same with sanitycheck.log.

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -p acrn --device-testing --device-serial /dev/ttyUSB1 -T tests
    --west-flash="acrn_up2.sh"
2. See error

**Screenshots or console output**
1. test results in acrn.xml
**testsuites><testsuite errors="49" failures="0" name="Sanitycheck" skipped="0" tests="51" time="1458.823267"**

2. test results in sanitycheck.xml
**testsuites><testsuite errors="0" failures="7" name="Sanitycheck" skip="20" tests="125" time="1458.823267"**

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 46330f13ed

nashif, 2020-03-09:
@chen-png this issue is very difficult to reproduce, I do not have the same setup with acrn in place, Do you get this issue with boards other than acrn?

chen-png, 2020-03-11:
>  Do you get this issue with boards other than acrn?

no, other boards are good.

chen-png, 2020-03-11:
for acrn, I found that in monitor_serial function, when invoking ser.readline() to get the serial output, the returned line value couldn't be correctly matched in test harness handle function, so sanitycheck couldn't get the pass/fail result foo each test unit, causing the generated acrn.xml error.

chen-png, 2020-03-11:
the normal output should like this, **the serial should read line by line**.(i add a print to serial)

DEBUG   - DEVICE: ===========================================
line: ================================================
DEBUG   - DEVICE: starting test - test_fifo_cancel_wait
line: starting test - test_fifo_cancel_wait
DEBUG   - DEVICE: PASS - test_fifo_cancel_wait
line: PASS - test_fifo_cancel_wait
DEBUG   - DEVICE: ======================================
line: ==============================================
DEBUG   - DEVICE: starting test - test_fifo_is_empty_thread
line: starting test - test_fifo_is_empty_thread
DEBUG   - DEVICE: PASS - test_fifo_is_empty_thread
line: PASS - test_fifo_is_empty_thread

**but for acrn output, it likes this, it didn't get the output line by line.**
DEBUG   - DEVICE: 
line:
starting test - test_fifo_cancel_wait
starting test - test_fifo_cancel_wait
PASS - test_fifo_cancel_wait
PASS - test_fifo_cancel_wait

starting test - test_fifo_is_empty_thread
starting test - test_fifo_is_empty_thread
PASS - test_fifo_is_empty_thread
PASS - test_fifo_is_empty_thread

chen-png, 2020-03-17:
for acrn, each serial line has a '\r' char in front of the output, likes this:
b'\r PASS - test_fifo_is_empty_thread \r\n'

after the decode('utf-8', 'ignore') and rstrip(), the result likes this:
   \r PASS - test_fifo_is_empty_thread

but this can't be match successfully in result_re.match in harness.py, this \r will influence the match result.
also in the generated sanitycheck.log, in each serial output, there is a ^M char.

so we should delete this \r char in serial output.

PR #23513 (2020-03-17): sanitycheck: delete the '\r' char in front of serial output
Add lstrip() function to delete the '\r' char in front of the
serial output from serail readline, because this '\r'
will influence the result match function in harness.py.

Fixes #23309
closed 2020-03-17 by nashif