zephyrproject-rtos/zephyr #23310: GUI: LVGL: possible NULL dereference

tluio, 2020-03-06:
Sorry, I am not sure if I can report bug in lvgl here, but I find issue may be not supported in zephyrproject-rtos/lvgl.

File: modules/lib/gui/lvgl/src/lv_draw/lv_img_decoder.c
Function: lv_img_decoder_built_in_open
Code:
391	       user_data->palette                         = lv_mem_alloc(palette_size * sizeof(lv_color_t));
392	       user_data->opa                             = lv_mem_alloc(palette_size * sizeof(lv_opa_t));
393 	       if(user_data->palette == NULL || user_data->opa == NULL) {
394	            LV_LOG_ERROR("img_decoder_built_in_open: out of memory");
395	#if LV_USE_FILESYSTEM
396	            LV_ASSERT_MEM(user_data->f);
397	#endif
398	        }
399	
400	        if(dsc->src_type == LV_IMG_SRC_FILE) {
401	            /*Read the palette from file*/
402	#if LV_USE_FILESYSTEM
403	            lv_fs_seek(user_data->f, 4); /*Skip the header*/
404	            lv_color32_t cur_color;
405	            uint32_t i;
406	            for(i = 0; i < palette_size; i++) {
407	               lv_fs_read(user_data->f, &cur_color, sizeof(lv_color32_t), NULL);
408	                user_data->palette[i] = lv_color_make(cur_color.ch.red, cur_color.ch.green, cur_color.ch.blue);
409	                user_data->opa[i]     = cur_color.ch.alpha;
410	            }
411	#else
412	            LV_LOG_WARN("Image built-in decoder can read the palette because LV_USE_FILESYSTEM = 0");
413	            return LV_RES_INV;
414	#endif
415	        } else {
416	            /*The palette begins in the beginning of the image data. Just point to it.*/
417	            lv_color32_t * palette_p = (lv_color32_t *)((lv_img_dsc_t *)dsc->src)->data;
418	
419	            uint32_t i;
420	            for(i = 0; i < palette_size; i++) {
421	                user_data->palette[i] = lv_color_make(palette_p[i].ch.red, palette_p[i].ch.green, palette_p[i].ch.blue);
422	                user_data->opa[i]     = palette_p[i].ch.alpha;
423	            }
424	        }
Description: On line 391, **user_data->palette** may be NULL, but on line 408 and 421, **user_data->palette** is dereferenced. So does **user_data->opa**. I guess LV_LOG_ERROR on line 394 does nothing but outputs the log. But I am not sure if **palette_size** must be 0 when **user_data->palette** is NULL.

nashif, 2020-03-06:
you should report those against lvgl, where those should be fixed

tluio, 2020-03-06:
Sorry. Thank you for your advice and now I think I should close this issue.

closed 2020-03-06 by tluio