zephyrproject-rtos/zephyr #23311: Sanitycheck flash error on frdm_k64f board.

chen-png, 2020-03-06:
**Describe the bug**
use sanitycheck script to run all tests on frdm_k64f board, sometimes it will meet a flash error.

it's ok to run the test build step, but when run flash step, **it  will failed occasionally**. 
the error log in device.log as below.

**To Reproduce**
Steps to reproduce the behavior:
1. sanitycheck -p frdm_k64f --device-testing --device-serial /dev/ttyACM1 -T tests/
2. See error

**Screenshots or console output**

**error log in device.log**
WARNING: runners.pyocd: hex file (/home/wentongw/work/testscript/sanity-out/frdm_k64f/tests/kernel/mem_protect/mem_protect/kernel.memory_protection/zephyr/zephyr.hex) does not exist; falling back on .bin (/home/wentongw/work/testscript/sanity-out/frdm_k64f/tests/kernel/mem_protect/mem_protect/kernel.memory_protection/zephyr/zephyr.bin). Consider enabling CONFIG_BUILD_OUTPUT_HEX.

0000278:WARNING:pyusb_v2_backend:[Errno 13] Access denied (insufficient permissions) while trying to interrogate a USB device (VID=0d28 PID=0204). This can probably be remedied with a udev rule. See <https://github.com/mbedmicro/pyOCD/tree/master/udev> for help.

0000286:CRITICAL:__main__:uncaught exception:
Traceback (most recent call last):
  File "/home/wentongw/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 338, in run
    self._COMMANDS[self._args.cmd](self)
  File "/home/wentongw/.local/lib/python3.7/site-packages/pyocd/__main__.py", line 468, in do_flash
    with session:
  File "/home/wentongw/.local/lib/python3.7/site-packages/pyocd/core/session.py", line 284, in __enter__
    **assert self._probe is not None**
**AssertionError
ERROR: command exited with status 1: pyocd flash -e sector -t k64f** /home/wentongw/work/testscript/sanity-out/frdm_k64f/tests/kernel/mem_protect/mem_protect/kernel.memory_protection/zephyr/zephyr.bin
make[3]: *** [zephyr/cmake/flash/CMakeFiles/flash.dir/build.make:58: zephyr/cmake/flash/CMakeFiles/flash] Error 1
make[2]: *** [CMakeFiles/Makefile2:3654: zephyr/cmake/flash/CMakeFiles/flash.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:3661: zephyr/cmake/flash/CMakeFiles/flash.dir/rule] Error 2
make: *** [Makefile:742: flash] Error 2

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 46330f13ed

nashif, 2020-03-06:
sanitycheck just uses west flash to flash a device, the fact that flashing fails is not related to sanitycheck. The flash failures is something that is common with the frdm_k64f for whatever reason, does not happen with other boards. In those cases sanitycheck can retry those failed tests multiple times until all flashing issues are corrected through retries leaving you with only the actual failures. 
If some boards keep failing to flash, this is most likely related to the board, the firmware/bootloader/debugger installed.

nashif, 2020-03-10:
works fine for me:

```
INFO    - 76 of 76 tests passed (100.00%), 0 failed, 7 skipped with 0 warnings in 251.92 seconds
INFO    - In total 94 test cases were executed on 1 out of total 230 platforms (0.43%)

Hardware distribution summary:

| Board     | ID                                               |   Counter |
|-----------|--------------------------------------------------|-----------|
| frdm_k64f | 0240000032044e4500097009997b00386781000097969900 |        41 |
| frdm_k64f | 0240000026334e450015400f5e0e000b4eb1000097969900 |        35 |
```

I had a  3rd board that was misbehaving and failing to flash, once I removed it, things pass with no flashing issues. Can you check your board please?

closed 2020-03-10 by nashif

chen-png, 2020-03-11:
yes, i will have a try.