zephyrproject-rtos/zephyr #23314: Bluetooth: controller: Use defines for 625 us and 1250 us

cvinayak, 2020-03-06:
**Is your enhancement proposal related to a problem? Please describe.**
Bluetooth controller implementation uses magic numbers 625 and 1250 for the advertising interval and connection interval units.

**Describe the solution you'd like**
Use defines for the values 625 and 1250

**Describe alternatives you've considered**
None

**Additional context**
Refer to comments in https://github.com/zephyrproject-rtos/zephyr/pull/21831

Nukersson, 2020-05-08:
Maybe Kconfig solution would be better?

cvinayak, 2020-05-11:
These are constant values, and used all around the implementation for varies purposes. Based on their use, suitable names need be given. If some of these are user (desired) configurables, then they can be Kconfig.

PR #31375 (2021-01-18): Bluetooth: Controller: re-factor magic constants
Defined 625 and 1250 as context specific interval common to ULL and LLL.
Checked with bluetooth sanity checks.

Fixes #23314.
commit b2fcda8e (2021-01-18): Bluetooth: Controller: re-factor magic constants

Defined 625 and 1250 as context specific interval common to ULL and LLL.
Checked with bluetooth sanity checks.

Fixes #23314.
closed 2021-01-18 by carlescufi