zephyrproject-rtos/zephyr #23319: hci interface stopped working after few hours/days

mayankpatel17, 2020-03-06:
Hi All,

I'm facing issues regarding ble's hci interface.
I'm having nrf52840 module on which i run zephyr's hci_uart app for beacon scanning.
I have Bluez 5.50 installed on my custom board (imx6ull).
i.MX6ULL is connected with nrf52840 chip over UART.
I'm using full UART (Tx, Rx, RTS, CTS). These 4 pins are connected & configured from both (i.MX6UL & nrf52840_pca10056) side.

Note : nrf52840 chip is flashed with HCI_UART sample app of zephyr. In that i have configured 4 pins and make HWFC enable from nrf5.conf file.

hci0 interface on my device is "UP RUNNING". So, i'm able to scan the beacons. (My application scans beacons after every 10 seconds).

Now, When i have put my device for over night testing, then i have observed that after few hours 
 (around 18-20) beacon scanning was stopped.
And i'm getting "Bluetooth: hci0 command 0x200b tx timeout" prints after every 5-10 seconds on the console, still the hci0 interface status is "UP RUNNING", but not able to scan the beacons.

Sometimes also getting the prints like "Bluetooth: hci0: Frame reassembly failed (-84)".

What could be the reason behind this ?
 
**To Reproduce**
- Run beacon scanning continuously (After gap of every 10-20 seconds) for a long time.

**Expected behavior**
- It should run continuously without having any problem.

**Impact**
This has a blocker kind of effect for my device (Because it stop scanning beacons and hence i'm not able to track the Assets).

Thanks,
Mayank

carlescufi, 2020-03-06:
* is there a way you can connect a logic analyzer to the pins to get the UART lines going in and out logged?  
* is there a way you can connect a debugger to the nRF52840 once it has stopped working to get a backtrace or some logging? A Segger debugger would be ideal, then we could use RTT

cvinayak, 2020-03-06:
Could you please provide details about how you build, and reproduction steps (command line or simple application that you can share), particularly if you have any changes to samples or its configuration. Also, any logs (console logs, sniffer traces, btmon logs etc... just before the issue). Environment details, commit hash, toolchain used.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=board\_xyz
3. make
4. See error

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

joerchan, 2020-03-06:
> is there a way you can connect a logic analyzer to the pins to get the UART lines going in and out logged?

This affects the UART lines I think. Might give it a try. But could end up it not being able to reproduce it if it is something in the flow-control I think.

carlescufi, 2020-03-06:
Relabeling as question because it's unclear whether this is a Zephyr bug or a board/tty issue.

carlescufi, 2020-03-06:
@mayankpatel17 please provide all the information required by @cvinayak in [this comment](https://github.com/zephyrproject-rtos/zephyr/issues/23319#issuecomment-595716997) and reply to the questions in [this other comment](https://github.com/zephyrproject-rtos/zephyr/issues/23319#issuecomment-595716943).

mayankpatel17, 2020-03-06:
> Could you please provide details about how you build, and reproduction steps (command line or simple application that you can share), particularly if you have any changes to samples or its [...]

I'm using zephyr's version 2.1.0. Compiling sample app using command "cmake -B build -DBOARD=nrf52840_pca10056 samples/bluetooth/hci_uart"

> **To Reproduce**
> Steps to reproduce the behavior:
> 
> 1. mkdir build; cd build
> 2. cmake -DBOARD=board_xyz
> 3. make
> 4. See error
>

To reproduce this just need to keep the scanning on for a long time. There is not any compilation issue i have faced.

> **Screenshots or console output**
> If applicable, add a screenshot (drag-and-drop an image), or console logs
> (cut-and-paste text and put a code fence (```) before and after, to help
> explain the [...]

Attaching file here which shows console logs before and after the issue.
[hci_stoppped_working_logs.docx](https://github.com/zephyrproject-rtos/zephyr/files/4298552/hci_stoppped_working_logs.docx)

-

mayankpatel17, 2020-03-06:
> * is there a way you can connect a logic analyzer to the pins to get the UART lines going in and out logged?
The whole connection between i.MX6ULL and nrf52840 is embedded together on board, i don't have any direct access to the pins. So I have no idea how i would be able to connect analyzer to read the UART lines !
> * is there a way you can connect a debugger to the nRF52840 once it has stopped working to get a backtrace or some logging? A Segger debugger would be ideal, then we could use RTT

I have nrf52840 embedded on my custom board. Still, i have connection for Jlink on board to the nrf chip whcih i'm using to flash nrf chip.

carlescufi, 2020-03-06:
> I'm using zephyr's version 2.1.0.

Would you mind trying with the current master branch and reporting whether you see the same symptoms please? There are some fixes since 2.1.0 that could apply to the issue you see.

mayankpatel17, 2020-03-09:
Trying to reproduce, will update here once anything is found.

carlescufi, 2020-03-10:
@mayankpatel17 were you able to reproduce with master?

mayankpatel17, 2020-03-11:
Hi @carlescufi ,
Not tested with the master, but currently trying to reproduce with the 2.1.0 zephyr version which i had used earlier for the build. Tried 2 times with overnight testing but not able to reproduce the issue. 
When i reported this issue, I had observed some UART console related errors in my board, then afterwards this Bluetooth related error started to came. So seems like issue was related to UART and may not be related to Bluez/zephyr/hci in general.

Let me try 2-3 more times (For 2-4 days), if then also i cannot produce the issue again then we can close this question.

Thank you,
Mayank

mayankpatel17, 2020-03-17:
Update : Trying with the zephyr V2.2.99, Facing some compilation issues.
I will update here once anything is found.

cvinayak, 2020-04-07:
@mayankpatel17 any updates?

mayankpatel17, 2020-04-10:
Hi Vinayak,

I have planned the retry with latest master branch but because of ongoing
nationwide lockdown I'm  not able to test it.
As soon as i will start or find anything on that part i will surely let you
know in this thread.

Regards,
Mayank Patel

carlescufi, 2020-07-09:
Closing due to inactivity, Please reopen if it's still an issue.

closed 2020-07-09 by carlescufi

avi951, 2024-04-15:
Still, I am facing this kind of issue for my board.
It shows me when I run **sudo hciconfig hci0 reset**

```
udo hciconfig hci0 reset
[ 1360.196739] Bluetooth: hci0: command 0x1003 tx timeout
[ 1362.329809] Bluetooth: hci0: command 0x1001 tx timeout
[ 1364.462769] Bluetooth: hci0: command 0x1009 tx timeout
```