zephyrproject-rtos/zephyr #23322: flash, spi-nor: Configuration of jedec spi nor flash device driver

rckTom, 2020-03-06:
**Is your enhancement proposal related to a problem? Please describe.**
With the current implementation of the jedec spi nor driver only devices are supported that have the ability to perform 4KB, 32KB, 64KB and chip level erase. But there are jedec compatible chips that do not support all of the mentioned erase types or at different granularity (e.g. Cypress s25fl512s just supports erase with a fixed size of 256KB). This can lead to memory corruption if one of these chips is used with the current driver. 

The jedec standard for serial flash discoverable parameters [1] specifies up to 4 erase types with the only requirement that the erase size has to be a power of two. The erase type specifies the size and the op-code. 

Besides the erase types, the SFDP contains other parameters like op-codes for power down, op-codes for advanced read and write features, supported addressing modes and op-codes to switch between them. These could also be integrated (related issue #22965 about 4 byte addressing). 

[1] [Serial Flash Discoverable Parameter, JESD216D.01](https://www.jedec.org/standards-documents/docs/jesd216b)

**Describe the solution you'd like**
As discussed on slack, a solution would be to provide all necessary parameters in a look up table for well known devices and fall back to the current default for generic ones. The compatible string from devicetree or the jedec id could provide the key for the look up table. Another solution would be to read the SFDP during startup form the chip, but as discussed on slack, this is not necessary for the devices targeted by zephyr. Maybe this could be an optional feature. 

**Additional context**
As mentioned above, there was a discussion about this issue on slack:

> Thomas Schmid 21. Feb. um 00:23 Uhr
> Hey. The current jedec spi nor flash implementation does not query the flash
> parameters during runtime but assumes some parameters like sector and page size as well as erase granularity and command op-codes. This can lead to some very hard to spot data corruption bugs if a flash device does not support all of the assumed erase commands (for example S25FL512S). The fix would be to query the parameters specified in JESD216 during startup and configure the driver accordingly. Before I dive into code: Is anyone already working on this?
12 Antworten

> carlesc  vor 15 Tagen
> But wouldn’t you know which spi nor flash is on your pcb? So that you can set the params correctly at build time? Or are you hot plugging flash chips?

> pabigot  vor 15 Tagen
> Reading the SFDP has been proposed, but as @carlesc points out isn't necessary for the devices Zephyr targets.  There's unlikely to be much interest in adding it.  If the devicetree binding doesn't allow you to accurately describe a device (e.g. 32-bit addressing) that'd be a different matter, but still not something that needs runtime SFDP queries to handle.

> Thomas Schmid  vor 15 Tagen
> Yes this is all static data and can be configured during compile time and i'm fine with that, but the point about the curent assumptions is still valid. There should be at least some more Kconfig options or devicetree configuration to support a broader set of jedec compatible chips.

> pabigot  vor 14 Tagen
> If there's a JEDEC capability we need that isn't supported by devicetree, please open an issue.  The only one recently has been 32-bit addressing.  AFAIK we've got standard erase block sizes covered.

> Thomas Schmid  vor 14 Tagen
> S25FL512S from cypress is jedec compatible and does only support sector erase with a sector size of 256Kbyte (no chip erase and no other erase size). If i read the current code correctly the driver assumes erase sizes of 64Kbyte, one sector erase with hard coded sector size of 4KByte and chip level erase. Only the 32Kbyte erase is optional.

> pabigot  vor 14 Tagen
> Yes, that seems to be very different from the devices encountered before.  There would have to be significant refactoring to support it, adding page, sector, and block sizes to the devicetree binding, along with individual control of whether the (formerly) standard erase commands are all supported.  The command that chip uses for sector erase is the command used by other devices for block erase.

> Thomas Schmid  vor 14 Tagen
> Yes. Furthermore the SPDF also specifies the commands for accesing the flash (read, write, erase). To be fully jedec compatible the driver would need the commands to be also configurable. This would be quite a list of options but making them optional with the current values as default should have minimal impact on the current users of the driver. I can work on the changes and provide a PR when i'm ready. Unfortunately one of my boards already depends on that chip and i have to find a fix for this.

> Serafin  vor 14 Tagen
> maybe have a config for chip "autodetect" and one for the current compilletime settings?

> brix  vor 14 Tagen
> One issue with the current approach is that it is very difficult to have second-source (multiple vendors) JEDEC SPI NOR flash devices. 

> pabigot  vor 14 Tagen
> We could add support to use the JEDEC ID to index into a table parameters for known devices, with the default being the existing assumptions about sizes and commands.  It would require a firmware update to support a new non-standard chip.  The table should probably be populated from content provided by devicetree so we don't waste flash on devices that will never be used in a particular application.

> pabigot  vor 14 Tagen
> We need an issue for this.  I'll open one later today and incorporate the discussion from this thread.

> brix  vor 14 Tagen
> Thanks.

carlescufi, 2020-03-06:
@rckTom would you consider sending a Pull Request with an initial implementation?

rckTom, 2020-03-06:
I'm working on it.

sslupsky, 2020-04-06:
@rckTom Would it be possible to add SPI NAND to this as well?  I've done some work to adapt the NOR driver to work with a Toshiba SPI NAND device.  I tried to make it as generic as possible but I wasn't able to find an JEDEC specification for SPI NAND that I could download.  I do know the Toshiba devices differ from the Winbond SPI NAND devices.

pabigot, 2020-08-24:
Closed by #25851.

closed 2020-08-24 by pabigot