zephyrproject-rtos/zephyr #23324: TinyCBOR is not linked to application files unless CONFIG_MCUMGR is selected

motet, 2020-03-06:
**To Reproduce**
This result can be reproduced by building the tests in the [COSE library](https://github.com/motet/cozy) I'm working on, and then subsequently removing the `select MCUMGR` line in the Kconfig file. This bug was pointed out on [Stack Overflow](https://stackoverflow.com/questions/59888006/building-tinycbor-module-with-zephyr) a month ago.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: latest Zephyr SDK
 - Zephyr 2.2.0-rc3

carlescufi, 2020-03-06:
Yep, will fix. But remember that this is quite a heavily modified fork of tinycbor and we plan to update it soon to align it with the upstream mcumgr one.

motet, 2020-03-06:
Excellent, thanks for the quick response.

de-nordic, 2020-03-09:
@motet can you please verify https://github.com/zephyrproject-rtos/zephyr/pull/23342 as the fix?

motet, 2020-03-09:
@de-nordic Done, everything works on my end. `tinycbor/cbor.h` gets linked with no errors now. Thanks!

de-nordic, 2020-03-09:
> 
> 
> @de-nordic Done, everything works on my end. `tinycbor/cbor.h` gets linked with no errors now. Thanks!

Thanks for checking that for me.

PR #23552 (2020-03-27): west.yml: Update revision of TinyCBOR
DEPENDS ON:
~~https://github.com/zephyrproject-rtos/tinycbor/pull/13~~
~~https://github.com/apache/mynewt-mcumgr/pull/74~~
~~https://github.com/zephyrproject-rtos/mcumgr/pull/18~~
~~https://github.com/JuulLabs-OSS/mcuboot/pull/688~~

Update provides fix for TinyCBOR include paths not being added into
compilation, unless MCUMGR has also been selected.

Fixes: #23324

NOTE: This PR replaces https://github.com/zephyrproject-rtos/zephyr/pull/23342, from branch incorrectly pushed to zephyrproject.
closed 2020-03-27 by carlescufi