zephyrproject-rtos/zephyr #23337: USB DFU device + Composite Device with ACM Serial - Windows Fails

optdcw, 2020-03-08:
I'm experiencing similar issue as this closed on [#15497] and [#14882]

I'm using USB DFU device with composite device ACM for serial port.
On linux everything runs ok, however on windows i keep getting the Lost device after RESET:
`dfu-util 0.9
Device really in Runtime Mode, send DFU detach request...
Resetting USB...

dfu-util: Lost device after RESET?`

The windows also expiriences issue with listing mentioned in: [#15497]
It is listing single partition.

The main clue is propably that windows DFU-util works when the composite device with USB ACM is disabled.

I also found pull request: #15818 which mentions that:
'dfu + composite mode ❌ (class_handler (and custom_handler) is hardcoded to primary set of descriptors, dfu has two set of descriptors and does not work in composite configuration)'

carlescufi, 2020-03-09:
@optdcw please provide more information, as requested in the GitHub issue template. At the very minimum wich board and Zephyr revision are you using.

optdcw, 2020-03-09:
Board: nrf52840 DK
Zephyr: 2.10

carlescufi, 2020-03-09:
> Board: nrf52840 DK
> Zephyr: 2.10

Thanks. For good measure, since several fixes have landed since 2.1.0 related to both CDC ACM and USB DFU, could you please try the current master and see if the problem is present there too?

emob-nordic, 2020-03-10:
@Which mcuboot verson have You used?

optdcw, 2020-03-10:
I will test it with the master branch until the end of the week.

I'm looking for using the dfu both inside mcu boot and in the user application to enable boatload during normal operations.
This issue was mainly in the user application.

emob-nordic, 2020-03-10:
What version of bootloader are You using?

optdcw, 2020-03-10:
I'm using fw-nrfconnect-mcu boot, I think version 1.4.0

jfischer-phytec-iot, 2020-03-11:
Please provide information about host OS version. Does the sample samples/subsys/usb/dfu work for you?

carlescufi, 2020-03-11:
@optdcw if possible, could you verify that the issue is also present in Zephyr 2.2 (released yesterday)?

optdcw, 2020-03-20:
The issue is indeed present in the Version Zephyr 2.2.
I have tried the MCU Boot example with this same result.
Log from MCU boot is following:
`00> *** Booting Zephyr OS version 2.1.99  ***
00> 
00> [00:00:00.000,000] <dbg> usb_device.usb_enable: lock usb_enable_lock mutex
00> [00:00:00.000,335] <err> usb_nrfx: nRF USB Attach
00> [00:00:00.000,701] <err> usb_nrfx: USB detected
00> [00:00:00.000,946] <err> usb_nrfx: HF clock start success (0)
00> [00:00:00.001,312] <err> usb_nrfx: EP enable: 0x00
00> [00:00:00.001,586] <err> usb_nrfx: EP enable: 0x80
00> [00:00:00.001,831] <dbg> usb_device.usb_enable: unlock usb_enable_lock mutex
00> [00:00:00.002,471] <inf> mcuboot: Starting bootloader
00> [00:00:00.002,868] <inf> mcuboot: Waiting for USB DFU
00> [00:00:00.005,065] <err> usb_nrfx: SUSPEND state detected
00> [00:00:00.005,371] <err> usb_[00:00:07.102,783] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:07.103,179] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:07.103,485] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:07.103,912] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:07.104,309] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:07.104,705] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:07.105,010] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:07.105,468] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.490,142] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.490,539] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:08.490,844] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:08.491,271] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.491,668] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.492,065] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x409
00> [00:00:08.492,401] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:08.492,828] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.521,850] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.522,216] <dbg> usb_device.class_handler: bRequest 0x3, wIndex 0x0
00> [00:00:08.522,552] <dbg> usb_device.class_handler: Handler 0x33fd, iface 0x0, index0x0
00> [00:00:08.523,071] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.796,417] <err> usb_nrfx: USBD reset event
00> [00:00:08.842,224] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.842,590] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:08.842,926] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:08.843,353] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.843,750] <err> usb_nrfx: USBD reset event
00> [00:00:08.885,284] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.885,650] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:08.885,986] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:08.886,413] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:08.886,810] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:08.887,207] <dbg> usb_device.custom_handler: bRequest 0x9, wIndex 0x0
00> [00:00:08.887,512] <dbg> usb_device.usb_handle_std_device_req: REQ_SET_CONFIGURATION, conf 0x1
00> [00:00:08.895,080] <err> usb_nrfx: SUSPEND state detected
00> [00:00:08.895,355] <err> usb_nrfx: USB Suspend state
00> [00:00:08.998,260] <err> usb_nrfx: RESUMING from suspend
00> [00:00:08.998,565] <err> usb_nrfx: USB resume
00> [00:00:08.998,809] <err> usb_nrfx: USBD reset event
00> [00:00:09.046,173] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:09.046,539] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:09.046,875] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:09.047,302] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:09.047,698] <err> usb_nrfx: USBD reset event
00> [00:00:09.090,515] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:09.090,911] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:09.091,247] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:09.091,644] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:09.092,193] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:09.092,559] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x0
00> [00:00:09.092,895] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:09.093,322] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
00> [00:00:09.094,421] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:09.094,818] <dbg> usb_device.custom_handler: bRequest 0x6, wIndex 0x409
00> [00:00:09.095,153] <dbg> usb_device.usb_handle_std_device_req: REQ_GET_DESCRIPTOR
00> [00:00:09.095,611] <dbg> usb_device.usb_handle_control_transfer: ep 0x80, status 0x02
[... 2 lines cut ...]
[... cut ...]
00> [00:00:09.134,246] <dbg> usb_device.usb_handle_control_transfer: ep 0x00, status 0x00
00> [00:00:09.134,613] <dbg> usb_device.custom_handler: bRequest 0x9, wIndex 0x0
00> [00:00:09.140,014] <dbg> usb_device.usb_handle_std_device_req: REQ_SET_CONFIGURATION, conf 0x1
00> [00:00:12.497,772] <inf> mcuboot: USB DFU wait time elapsed
00> [00:00:12.498,565] <inf> mcuboot: Primary image: magic=good, swap_type=0x4, copy_done=0x1, image_ok=0x1
00> [00:00:12.498,992] <inf> mcuboot: Boot source: none
00> [00:00:12.499,298] <inf> mcuboot: Swap type: none`

emob-nordic, 2020-03-23:
I am experiencing issue with dfu-util. I think its because of linux vm. I will try to help as soon i get done with that.

emob-nordic, 2020-03-23:
It happened only on windows, right? I think the problem is with IAD descriptor which is not present when You config ```CONFIG_USB_COMPOSITE_DEVICE=y``` only. I will have a deeper view on this.

optdcw, 2020-03-23:
Yes it happends only on windows.

emob-nordic, 2020-03-24:
Could You send me prj.conf that You are using for build the dfu + cdc sample?

optdcw, 2020-04-02:
So i made clean installation of zephyr 2.2.0.
I have build example ./samples/usb/cdc/ whiched worked.
Then i aplied the overlay-composite-cdc-dfu.conf and the application did not insert any usb.
I noticed that the application build with the overlay is shifted and starts at 0xC000 instead of 0x0.

Which is the reason i guess the example is not working. It propably expects mcuboot to be there however it is not. Which is strange given that the overlay has the CONFIG_BOOTLOADER_MCUBOOT=y in it.

The complete prj.conf for the example is [prj.txt](https://github.com/zephyrproject-rtos/zephyr/files/4423628/prj.txt)

optdcw, 2020-04-02:
So i made the similar test with nrf connect sdk zephyr version: Zephyr version: 2.1.99.
There the mcuboot created merged.hex as expected. 

The application booted sucessfuly. I used this [prj.conf](https://github.com/zephyrproject-rtos/zephyr/files/4423687/prj.txt).

I installed the driver using zadig: WinUSB (v6.1.7600.16385).

The output of the dfu util is here [dfu_update.txt](https://github.com/zephyrproject-rtos/zephyr/files/4423707/dfu_update.txt)
Log from the application: [log_nrf52.txt](https://github.com/zephyrproject-rtos/zephyr/files/4423724/log_nrf52.txt)

optdcw, 2020-04-21:
Any progress regarding this issue ?

emob-nordic, 2020-04-21:
Together with @carlescufi and @jfischer-phytec-iot  we came up with conclusion:
Windows can handle dfu devices **only** when they are standalone (nothing more except dfu. No more functions like cdc/hid/whatever). It's not possible for windows (its not stable) to 'talk' to both cdc+dfu at the same time. Its the host flaw. We do not know how to come around this at the device level.

For me it happend once that I could actually use cdc+dfu on windows but this required swapping the drivers from one to another several times. It's very unstable. Why don't You configure dfu to actually flash the device with cdc function?

I was about to refer to this issue by preparing the patch to the sample doc that actually using dfu+cdc on Windows may not work and We do not recommend using this configuration.

optdcw, 2020-04-21:
I tried using the cdc-serial bootloader and got it to work however the dfu worked faster and seemed much more robust. 

If you don't recommend using this kind of configuration i will revert to using the cdc-serial bootloader. Which is a little bit unconvinient.

emob-nordic, 2020-04-23:
Hi, I think under the circumstances we have this would be necessary for You to use cdc-serial instead of dfu.

jfischer-phytec-iot, 2020-04-23:
@optdcw To clarify, is it really necessary to trigger update process from application side without any barrier? 
I am not familiar with Windows OS driver model, but for example if you have a device with composite support for DFU + CDC ACM, and dfu-util triggers a download or upload, the device would re-attach in update mode without CDC ACM interface. As there is no barrier, any app/user with sufficient rights could break application which is using CDC ACM interface.

optdcw, 2020-04-23:
@jfischer-phytec-iot we are currently in development stage of protable device, which will have usb interface only for chargin and bootload/debuging purposes. So it would be easier for us to use existing software to perform the DFU without the need to switch between them using CDC ACM. Which could be simple command.

Your idea is also interesting to try, however it would require to modify existing usb subsystem parts to enable such behavior. Or is there some support for switching between usb interfaces ? 

The other option is to have USB DFU enabled inside MCUboot which would then wait for usb dfu. However i have not seed option for applicatino to specify for the mcu boot to wait for host after next reboot(which would be very convinient function).  And the main application would after boot command restarted for the mcuboot bootload.
The mcuboot wait for dfu with timeout is very unconvinient because it delays the start of the whole device.

carlescufi, 2020-05-06:
@jfischer-phytec-iot if you have a chance, can you please comment on the question from @optdcw?

jfischer-phytec-iot, 2020-05-06:
> @jfischer-phytec-iot we are currently in development stage of protable device, which will have usb interface only for chargin and bootload/debuging purposes. So it would be easier for us to use [...]

I suggest to go with two CDC ACM interfaces (one for update and the other for logging) for now.

> Your idea is also interesting to try, however it would require to modify existing usb subsystem parts to enable such behavior. Or is there some support for switching between usb interfaces ?

No, the stack supports only one configuration. I have an idea how to rework the initialization/interface of the classes so that the user or an application could initialize a class on demand and then enable usb stack, but it would take a while to get it mainline.

> The other option is to have USB DFU enabled inside MCUboot which would then wait for usb dfu. However i have not seed option for applicatino to specify for the mcu boot to wait for host after next [...]

I do not know if there is a way to trigger an user defined behavior in MCUboot from the application.

> The mcuboot wait for dfu with timeout is very unconvinient because it delays the start of the whole device.

emob-nordic, 2020-05-14:
Hi. I gave myself very last chance to this issue and tried to make it work with windows as well. From what I've found out it's impossible and and I would like to present a nice explanation why.

I've started with pure bootloader installed on the nRF52840dk_nrf52840.
```~/zephyr/bootloader/mcuboot/boot/zephyr$ west build -b nRF52840dk_nrf52840 ``` ```west flash```

emob-nordic, 2020-05-14:
Hi,
I gave myself very last chance for this particular issue and tried to make it work.
After what I've tried to do I came up some conclusions I would like to share
to make it bright and clear why I think for this we cannot do much and this will not work.

Prepare
-

Build and flash the bootloader:
```~/zephyr/bootloader/mcuboot/boot/zephyr ```
Then dfu+cdc sample:
```~/zephyr/zephyr/samples/subsys/usb/cdc_acm/```
 with proper overlay:
 ``` west build -b nrf52840dk_nrf52840 -- -DOVERLAY_CONFIG=overlay-composite-cdc-dfu.conf```
 This application must be signed in order for bootloader to properly load the image ([instruction](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/usb/dfu)).
 Flash using ```west flash --hex-file signed.hex```
 
Then prepared signed bin file for dfu upload. I personally took blinky. Simialr steps as with cdc+dfu app expect signed bin is required.

General Notes
-

Zephyr is linnking descriptors in special RAM seciton and does it by sorting descriptors aplhabetically.
In our case First 2 interfaces is cdc_acm funciton and last is dfu interface.

[image]

Dfu device contains of 2 descriptor sets:
- Run-Time Descriptor set
- DFU Mode Descriptor set

Dfu device is returning Run-Time descriptor when 1st connected to the Host.
On special Host request (dfu-util here) the device is changing its descriptor set to DFU mode
where download/upload can be performed.

Linux testing
-

Starting from APPEND in the screenshot is example how linux handles dfu capable device.
Note wIndex = 0x02 meaning that interface 2 (dfu) is addressed.

[image]

Everything works fine there and image is downloaded correctly.

Windows testing
-

For Windows You must install driver by Your own. At the first attachement of the composite device
Windows does not attach any driver to the Interface 2 (dfu).

[image]

Note the driver is *NONE*.
I've installed the driver manually using Zadig v2.5.0.

Here is what happend when I tried the dfu download.

[image]

Treat 'APPENDs' as delimiters for dfu download tries.

I am enclosing output form cmd window.

```
Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0001
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Resetting USB...
Lost device after RESET?

C:\Users\emob\Downloads\dfu-util-0.9-win64>dfu-util-static.exe --alt 1 --download signed-zephyr.bin

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Invalid DFU suffix signature
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0001
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: Device does not implement get_status, assuming appIDLE
Device really in Runtime Mode, send DFU detach request...
error detaching
Resetting USB...
Lost device after RESET?

C:\Users\emob\Downloads\dfu-util-0.9-win64>  
```

At the very first attpemt Windows is addresing proper interface wIndex = 0x02.
This seems okay but it never resets the device and because of that dfu is not able to switch its descriptor to dfu mode.

Second attempt fails at requests because the device has already swapped
to DFU-Mode descriptors and expected the Host to reset the device.

The problem is: Windows never resets the device after requesting to switch to DFU-mode.

Sorry for long post, here is a potato.
[image]

PR #25347 (2020-05-27): usb: doc: Add note about cdc+dfu Windows OS exception.
Add a note for composite (CDC+DFU) device overlay.
Composite device CDC+DFU may not work with Windows OS host.
Windows OS does not send reset after DFU_DETACH request
(does not re-enumerates) and thus make it unable for
the device to restart in DFU mode.

Closes #23337.
closed 2020-05-27 by carlescufi