zephyrproject-rtos/zephyr #23346: bl65x_dvk boards do not reset after flashing

lairdjm, 2020-03-09:
**Describe the bug**
After flashing bl65x_dvk boards, they do not automatically reset

**To Reproduce**
Steps to reproduce the behavior:
ninja flash

**Expected behavior**
The module to reset

**Impact**
Slight annoyance

lairdjm, 2020-03-09:
Fix for 1.14: https://github.com/zephyrproject-rtos/zephyr/pull/23344
Fix for master: https://github.com/zephyrproject-rtos/zephyr/pull/23345

closed 2020-03-10 by jhedberg

lairdjm, 2020-03-10:
There is still https://github.com/zephyrproject-rtos/zephyr/pull/23344 which needs to be merged (on the 1.14 branch)

jhedberg, 2020-03-10:
> There is still #23344 which needs to be merged (on the 1.14 branch)

I've added it to the "To-do" column of https://github.com/zephyrproject-rtos/zephyr/projects/19 so that it gets processed appropriately.

PR #23345 (2020-03-10): boards: arm: bl65x_dvk: fix reset not working
Switch from using pin reset to normal reset.

Fixes #23346
PR #23344 (2020-03-18): boards: arm: bl65x_dvk: fix reset not working
Switch from using pin reset to normal reset.

Fixes #23346