zephyrproject-rtos/zephyr #23347: net: ieee802154_radio: API improvements

rlubos, 2020-03-09:
Hi,

As we're now putting more effort into 802.15.4 based protocols (extending OpenThread support, trying to run Zigbee over 802.15.4 radio driver in Zephyr), we've identified a need for several API improvements. Before pushing PRs with actual changes, I wanted to create this umbrella issue, summarizing the changes needed with some explanation. Hopefully, we can get some early feedback here.

### 1. Specify TX mode on transmission

While radio driver API has means to indicate CSMA CA support (`IEEE802154_HW_CSMA`), there's actually no way to indicate when CSMA CA procedure should be used. We have cases, when we need to support both, TX preceded with CSMA CA as well as TX w/o CCA and currently there is no way to choose the TX mode.

I propose to extend the `tx` API with an extra parameter, indicating TX mode. All existing `tx` calls would assume direct transmission (no cca). If an appropriate capability is returned by the radio driver, additional modes could be used. Besides CSMA CA mode, we also have a need for delayed transmission (handled at radio driver level), for that case `net_pkt->txtime` parameter could be used. Here's a summary of the changes proposed in the tx function:
https://github.com/rlubos/zephyr/commit/a6e90c9ceae420e8d2dd50b894f5943b59167ce6

### 2. Unify configuration API

Besides a "setter" API for configurations, our Zigbee library has a need to read configurations back. Of course, this could be handled in some intermediate layer, but I believe a neater approach would be to provide a "getter" function for configs in the API. I propose to take the same approach as ethernet driver API, where `set_config` and `get_config` functions are available:
https://github.com/zephyrproject-rtos/zephyr/blob/master/include/net/ethernet.h#L240

Optionally, we could also unify older API functions (`set_channel` and `set_txpower`) with `set_config` API. A summary of the proposed changes can be viewed in the commit:
https://github.com/rlubos/zephyr/commit/873c7b8dd37e0591fa0ff38b2137cef2e2f6be51

### 3. New configuration options

We've identified a need for new configuration options:
	- Promiscuous mode
	- PAN coordinator
	
Promiscuous mode is needed by both, Thread and Zigbee. There is a capability to indicate promiscuous mode support (`IEEE802154_HW_PROMISC`) but there's actually no way to put the radio into this mode. Optionally, we could consider putting the radio into promiscuous mode in case no filters are configured, but IMO it's simpler just to have a single switch.

Zigbee has to support PAN coordinator role, at least for nRF we need to set this role in the radio driver. A summary of the changes:
https://github.com/rlubos/zephyr/commit/b6b59e299aa3aee4b88da323293fd734e00885ce
	
### 4. Event notification 

OpenThread expects `otPlatRadioTxStarted` being called when the actual transmission starts. With CSMA/CA, this can be problematic as the `tx` function finishes only after the transmission is done (or failed).
That's why we propose to add a possibility to register an event notification function so that radio driver could notify the upper layer about certain events in an asynchronous way. The function could be registered with the existing `configure` API:
https://github.com/MarekPorwisz/zephyr/commit/1548e093d62983a1b838e29c94ee4fabbe643d13

	
### 5. Other minor changes:

- Remove OT conditional from Energy detection (as it's used by Zigbee as well),
- FCB information on RX (notify the upper layer if the radio driver set the Frame Pending Bit in the ACK or not),
- FCB mode (there are slight differences between Thread/Zigbee in Frame Pending Bit handling)
	
Here's a branch with overall changes: https://github.com/rlubos/zephyr/commits/ieee802154/radio-driver-api-extension

Please, let me know if you see any major blockers for the proposal above, or see that some of the changes proposed should be handled the other way. If there are no complaints, I can start posting pull requests with the API rework.

rlubos, 2020-03-09:
CC @tbursztyka @jukkar @MarekPorwisz @mariuszpos

jukkar, 2020-03-09:
LGTM. I didn't spot any big issues with these API changes. Most of the changes seem to be around get/set config, are you planning to send PR that implements these in some driver?

rlubos, 2020-03-10:
@jukkar Yes, the plan is to implement these changes step by step in nRF5 radio driver (and of course rework other radio drivers as needed).

tbursztyka, 2020-03-10:
@rlubos I am also fine with these changes (see below for details). However, do each and every changes iteratively (don't mix promiscuous and pan coordinator for instance, have one patch for each), also for each and every feature added: upgrade the drivers and the L2(s) as well. 

- For the config stuff, I am not entirely sold on the get_config part. To me, the driver should respond to the L2 part which owns the logic, and usually it's less hassle and more logical to store config bits and pieces in some context (see include/net/ieee802154.h, there is already such config holder), handled by the L2. Less repetitive code (each and every driver do not have to implement a get_config and the means to store such config etc...) and less calls as well. Generic behavior can be centralized in L2(s).
- Yes, have a switch for promiscuous. There was a plan to support it, but indeed besides the caps, it never went further.

nandojve, 2020-03-10:
I found by coincidence this topic and I would like comment two things:.

**Assyncronous TX**
When I implemented RF2XX driver I'm missed asynchronous TX behavior. Something like the upper layer start TX, driver program TX and immediate return. Then when radio finish TX a callback is executed passing only TX status, something like:
```
TX INVALID
ACCESS FAILED (CSMA failed)
NO ACK RECEIVED
TX OK (valid ACK received or NO ACK needed)
TX DP (idem TX OK + data pending bit)
```
I believe L2 can benefit with that behavior and than any logic related to CSMA/CA can be programmed at TX request following any standard, if it is the case.

One important thing (at least on RF2XX), it have normal and enhanced mode. On enhanced mode transceiver execute all CSMA/CA and re-transmissions without MCU intervention. This is very important on Low Power Networks and it will be sad lost that capability. At end, if we can select that behavior at TX it will be great.

**Low Power Mode**
Simple API to put node on LPM to save power, a On/Off parameter will be enough.

tbursztyka, 2020-03-10:
@nandojve isn't the proposed '**4. Event notification**' solving your asynchronous TX request?

nandojve, 2020-03-10:
@tbursztyka, yes sure, I only reinforced what I saw and what can have like enhanced Tx mode and LPM. I'm perfectly fine with Event notification.

nandojve, 2020-03-11:
I'm review pending issues and this #23193 reminds me that we need improve a way to provide MAC address.

rlubos, 2020-03-11:
Thank you all for the feedback,

@tbursztyka 
I see your point on `get_config`. For us it was more of a nice-to-have feature for now as, just as you say, L2 can take responsibility of storing the configuration. Let's leave it as it is for now.

@nandojve 
Async API feels more natural for us as well. Both nRF5 radio driver and OpenThread radio APIs are asynchronous, so right now we need a separate thread to simulate this behaviour.

We could add a new capability, `IEEE802154_HW_ASYNC_TX` which would indicate asynchronous behaviour, and then use the event handler as suggested.

What I'm concerned about though is that all existing L2s would need to support both, sync and async APIs for backward compability. I'm not bold enough to re-work all existing radio drivers, as it is a complex change.

@jukkar 
I have not found an `experimental` tag for the 802.15.4 driver/subsystem. We can limit the API changes to a single change in the `tx` function (new parameter). How should this work, do we need some deprecation procedure, like marking the exsiting `tx` as `tx_legacy` or the new one as `tx_new`?

rlubos, 2020-03-11:
@nandojve 
As for the low power mode, isn't the existing `start`/`stop` enough?

tbursztyka, 2020-03-11:
I guess it's ok to push changes without deprecating current API, or then it will quite painful.

Some devices does not support such async tx stuff. So depending on IEEE802154_HW_ASYNC_TX, l2 will have to know what to do (sync or async). You need to support both. Also, if a device supports async tx, it's because it can run the radio strategy by itself as well (aloha, csma-ca, etc...), such capabilities which will also be exposed.

rlubos, 2022-02-25:
The issue is outdated, closing.

closed 2022-02-25 by rlubos