zephyrproject-rtos/zephyr #23349: Question: How to add external soc, board, DTS, drivers and libs?

Aukstkalnis, 2020-03-09:
Hi.

I am trying to add ST's BlueNRG-2 soc to my project. Since zephyr does not have it, i want to add it to separate directory. I know there is such things as BOARD_ROOT, SOC_ROOT and DTS_ROOT. As I understand, you can set separated repositories for Board, soc and DTS. And what to do about other repositories, like Include, drivers, etc. There is no environment variable like SOC_ROOT. Where to add these?

carlescufi, 2020-03-09:
Hi @Aukstkalnis you can add out-of-tree drivers just as easily, here's a starting point:

https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/application_development/out_of_tree_driver

Your out-of-tree driver doesn't need a `DRIVER_ROOT` because it will be part of the build as soon as you add a Kconfig entry for it, even if that is outside the main zephyr tree, it's all part of `ZEPHYR_BASE` in this case.

Aukstkalnis, 2020-03-10:
Ok, I will try this. But there is one thing that is not clear to me. Its about where should the all things related soc be placed. In this example, it is placed directly in zephyr source tree. What if I want to hold soc related stuff (including drivers, libs, etc.) in one directory, board in another, app in another directory and zephyr also in another? If I use west, then it download everything to zephyraproject directory. Should I add soc and board to app directory as git submodule or somehow I can use west to manage app folder?

SebastianBoe, 2020-03-10:
It is up to you if you want west to manage your app directory or not. I can't tell you what you should do, because this depends on many things.

But I can explain what is supported to do.

https://docs.zephyrproject.org/latest/application/index.html#creating-an-application
https://docs.zephyrproject.org/latest/guides/west/index.html

Aukstkalnis, 2020-03-12:
Ok. Thanks.

Now I am trying to attach drivers. But I down understand how to do it. I aplication developement example "out_of_ tree_driver" shows how to define driver that is not related to soc. But I need to define GPIO, Clock, Pinmux, etc. that are soc's setup. Do I need to use some principal as external module? Maby it will be more clear if i just show you what I am trying to do :) https://github.com/Aukstkalnis/zephyr-app

Aukstkalnis, 2020-03-12:
I managed to fix problems with DTS, at least for now. Now I am stuck with SOC drivers... I am not sure where they should be placed and how to address them as if they where in $ENV{ZEPHYR_BASE}/drivers directory.

SebastianBoe, 2020-03-16:
You can put the SOC drivers in a "Zephyr module".

Here is one way of adding a zephyr module: https://github.com/zephyrproject-rtos/zephyr/blob/master/samples/application_development/out_of_tree_driver/CMakeLists.txt#L5

For other mechanisms, see https://docs.zephyrproject.org/latest/guides/modules.html

Aukstkalnis, 2020-03-24:
@SebastianBoe thanks for the tips. Zephyr now compiles and runs on blueNRG-2 SoC :) But not everything are clear. I would like to know what to do with Kconfig. Now I have a problem, that console is not working. UART works with uart driver, but not printk... Before compiling I get these warnings:
warning: UART_CONSOLE (defined at drivers/console/Kconfig:47) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: SERIAL_HAS_DRIVER (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_CONSOLE.html and/or look up
UART_CONSOLE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: The choice symbol SHELL_LOG_LEVEL_INF (defined at
subsys/logging/Kconfig.template.log_config:17) was selected (set =y), but no symbol ended up as the
choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_SHELL_LOG_LEVEL_INF.html and/or look
up SHELL_LOG_LEVEL_INF in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

I don't understand why. And how I can integrate my driver's configuration into Kconfig like they were in ZEPHYR_BASE source tree?

Aukstkalnis, 2020-04-24:
@tejlmand will you help?

tejlmand, 2020-07-01:
@Aukstkalnis sorry for late response, somehow this question never caught my attention.
If you have created the driver as a Zephyr module: https://docs.zephyrproject.org/latest/guides/modules.html#module-inclusion
then you can add your Kconfig file into the build system, and it should show up in the Kconfig menu, for example:
```
build:
  cmake: .
  kconfig: Kconfig
```

tejlmand, 2020-08-06:
@Aukstkalnis I hope the answers has helped you.
If you still need assistance, feel free to reopen the question.

closed 2020-08-06 by tejlmand

Aukstkalnis, 2021-08-13:
Sorry about late response. I could not make bluenrg soc to work out-of-tree so i and my colleague decided to fork zephyr and just add new soc.