zephyrproject-rtos/zephyr #23360: pissible NULL dereference in  zephyr/arch/arm/include/aarch32/cortex_m/exc.h

tluio, 2020-03-10:
I think there may be a null-pointer-dereference bug. I am not sure if it has something to do with [#17656](https://github.com/zephyrproject-rtos/zephyr/issues/17656)

File: 
       zephyr/arch/arm/core/aarch32/fatal.c
       zephyr/kernel/fatal.c
       zephyr/arch/arm/include/aarch32/cortex_m/exc.h
Function:
       z_arm_fatal_error
       z_fatal_error
       arch_is_in_nested_exception

Code:
       zephyr/arch/arm/core/aarch32/fatal.c     z_arm_fatal_error
41        void z_arm_fatal_error(unsigned int reason, const z_arch_esf_t *esf)
42        {
43
44	           if (esf != NULL) {
45		         esf_dump(esf);
46	           }
47	           z_fatal_error(reason, esf);
48          }

       zephyr/kernel/fatal.c        z_fatal_error
114  #if defined(CONFIG_ARCH_HAS_NESTED_EXCEPTION_DETECTION)
115	             if (arch_is_in_nested_exception(esf)) {
116		             LOG_ERR("Fault during interrupt handling\n");
117	             }
118 #endif

       zephyr/arch/arm/include/aarch32/cortex_m/exc.h      arch_is_in_nested_exception
71          static ALWAYS_INLINE bool arch_is_in_nested_exception(const z_arch_esf_t *esf)
72          {
73	             return (esf->basic.xpsr & IPSR_ISR_Msk) ? (true) : (false);
74          }
Description: On line 47 in function z_arm_fatal_error, **esf** may be NULL. Then it is passed to function z_fatal_error as a parameter. In z_fatal_error, arch_is_in_nested_exception is called with the argument **esf**. Finally, in function arch_is_in_nested_exception, **esf** is dereferenced on line 73, but now it may be NULL.

ioannisg, 2020-03-10:
There is a bug-fix PR for this, see #23356

tluio, 2020-03-10:
Ok, thank you.

ioannisg, 2020-03-10:
@tluio , I re-opened this; it will be closed when the bug-fix PR is merged.
Thanks for reporting it, by the way.
Would you mind taking a look at #23356 and see if this addresses your issue?

tluio, 2020-03-10:
> Would you mind taking a look at #23356 and see if this addresses your issue?

Yes, this PR addresses my issue. Thank you!

PR #23356 (2020-03-11): Arch arm cortex m spurious fix
- We fix the behavior of the spurious interrupt handler for ARM Cortex-M:
The behavior is now _identical_ to all other architectures (maybe except for ARC; FYI @vonhust ):
tjhe spurious ISR handler calls the z_arm_fatal_error(K_ERR_SPURIOUS_ISR, NULL).

The fix is done for Cortex-M but it affects shared code with Cortex-R (@stephanosio FYI)
- We fix the documentation of z_irq_spurious() for ARM, stressing that it is installed in _sw_isr_table (that can  have dynamic IRQs or not).

- We fix the *kernel*  z_fatal_error() implementation specifics for ARM so it check [... cut]
closed 2020-03-11 by jhedberg