zephyrproject-rtos/zephyr #23367: GPIO - Add different callbacks per trigger type

sparkhead95, 2020-03-10:
**Is your feature request related to a problem? Please describe.**
Taking buttons as an example - currently the user configures the GPIO port device with what triggers interrupts for the given GPIO pin, then adds a callback method for that pin using the callback structure. 
If I want to detect button down and button up, I have to read the GPIO pin inside the callback to figure out what state the button was when it was triggered - which is consuming interrupt time and power.

**Describe the solution you'd like**
Instead of defining what triggers the interrupt at the `gpio_pin_configure` (or more recently, `gpio_pin_interrupt_configure`), perhaps add the custom callback to the `gpio_pin_interrupt_configure` API call. This will allow a trigger type to be specified, and a given callback for this trigger type. 

**Describe alternatives you've considered**
(As described in the first section)

pabigot, 2020-03-10:
IMO the biggest problem with "trigger on both edges, sample in handler" is not the inconvenience but the inaccuracy: you don't know how the value read within the handler corresponds to the value that triggered the interrupt.  Consequently you can mis-identify the current state of the system.  My preferred solution is to trigger on a single edge, and within the handler reconfigure for the other edge, testing to be sure the sampled edge is consistent with the desired trigger direction.  (An implication of the feature description is that the hardware could tell you which edge was triggered, which I suspect is generally not the case.)

#17102 proposed some changes to the GPIO interrupt configuration that might help witih this but it got no traction.  I'm not sure if the current thinking is to support multiple callbacks for a single pin; my understanding from discussion during the GPIO rework was that some people thought that was not necessary.

sparkhead95, 2020-03-10:
> #17102

Ah I didn't realise you could reconfigure the interrupt inside the interrupt handler! I suppose I can do that for now - I did notice that while testing it the current way, some results were inconsistent.

I know it's probably not my place to say but if we're both using workarounds for this, surely that deems it necessary?

pabigot, 2020-03-10:
It's extremely important to be able to reconfigure the interrupt within the handler.

> I know it's probably not my place to say but if we're both using workarounds for this, surely that deems it necessary?

Depends on how much you expect the OS to do for you.  One perspective is Zephyr should provide all the functionality necessary so applications can be written that will work on all platforms without modification. In that view a generic solution is useful.

Another is "I want the OS to provide me basic services and the hard stuff like Bluetooth, but otherwise get out of my way so I can leverage the capabilities of the hardware I know I'm using".  Which is where I generally come from.  In my own work I don't generally even use the vendor HALs; I'd rather manipulate peripherals directly through pointers to CMSIS-defined register layouts, so I don't have to keep looking into what the intermediate functions are doing.  Sometimes this ends up a bad decision, but mostly it works out quite well.

All of which is beside the point regarding that issue: there wasn't enough interest in reworking interrupts to make progress.  This issue might increase such interest.