zephyrproject-rtos/zephyr #23387: [Question] Why does not zephyr use a toolchain file with cmake as -DCMAKE_TOOLCHAIN_FILE=.. ?

trondhe, 2020-03-11:
I just wondered why Zephyr does not use CMake's recommended way of cross-compiling by calling it with a toolchain file on the command line as

    cmake  -B buildfolder -S sourceFolder -GNinja -DBOARD=nrf52840_mdk -DCMAKE_TOOLCHAIN_FILE=somePath

but instead just use an include statement at the top of the project?
What is the motivation and arguments for/against doing it this way?

carlescufi, 2020-03-11:
One of the requirements when we transitioned from make/Kbuild to CMake was to be able to do something like:

`cmake -GNinja -DBOARD=<board> ..`

and everything should magically work. Now that west is so pervasive we could reconsider this.

trondhe, 2020-03-11:
In my experience it couples the sdk harder into the project using it more than necessary. A bit of the same error Nordic made with their mesh sdk, though zephyr has implemented it in a much nicer way.

Zephyr seems to be nicely laid out to be able make this change as everything is built as libs and linked in to a final product.

SebastianBoe, 2020-03-11:
When migrating to CMake, we were competing against a CLI that looked like:

make

To be competitive, we could not require users to specify the toolchain file on every CMake invocation. Note that west did not exist at the time.

tejlmand, 2020-03-11:
> To be competitive, we could not require users to specify the toolchain file on every CMake invocation. Note that west did not exist at the time.

I don't see @trondhe requesting us to remove this feature.
He is just asking that it is also possible to simple use a CMake Toolchain file, for users prefering this.
And to experienced CMake users, I think it is valuable if we consider how both principles can be supported, with Zephyr recommending the current solution.

trondhe, 2020-03-11:
> I don't see @trondhe requesting us to remove this feature.
> He is just asking that it is also possible to simple use a CMake Toolchain file, for users prefering this.

Yes and no I guess.
If the final implementation that supports both including it with `include($ENV{ZEPHYR_BASE}/cmake/app/boilerplate.cmake NO_POLICY_SCOPE)` and as a direct toolchain file on the CLI is inferior to a solution that is exclusive or, then a decision should be made.  

If however supporting both ways plays out well both in as interface for users and maintainability, then that is of course a good option.

nashif, 2020-03-11:
IMO we should reconsider this and add support for toolchain files, IMO this is orthogonal to west and should be kept orthogonal to west. Integration with toolchains that already support CMAKE and integration with IDEs that expect a toolchain file makes this necessary. Ease of use with support toolchains and zephyr SDK is a nice thing, but we cant expect every toolchain to be added the same into Zephyr...

SebastianBoe, 2020-03-11:
A prototype that proves that supporting toolchain files will make integration easier is welcome.

But until proven otherwise I am sceptical and would not recommend investing time into this.

SebastianBoe, 2020-03-27:
The question of why we don't use a toolchain file is answered.

closed 2020-03-27 by SebastianBoe

KKoovalsky, 2021-10-31:
What is the status on this? Are there any plans to support `CMAKE_TOOLCHAIN_FILE`?

SebastianBoe, 2021-11-01:
No