zephyrproject-rtos/zephyr #23393: ELF file size with code coverage functions is more than FLASH space

kkiyota, 2020-03-11:
**Is your enhancement proposal related to a problem? Please describe.**

In case that I enabled CONFIG_COVERAGE=y as shown below, error occurred because ELF size was increased a lot compared to disabling CONFIG_COVERAGE which is 26208 B.  

> cmake -GNinja -DCONFIG_COVERAGE=y ..

> Memory region         Used Size  Region Size  %age Used

           FLASH:      217176 B       128 KB    165.69%
            SRAM:      113068 B       832 KB     13.27%
        IDT_LIST:         216 B         2 KB     10.55/home/koji/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: zephyr/zephyr_prebuilt.elf section `rodata' will not fit in region `FLASH'
/home/koji/Downloads/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 86104 bytes
collect2: error: ld returned 1 exit status
%
ninja: build stopped: subcommand failed.

**Describe the solution you'd like**

Are there any way to specify the code coverage functions into SRAM space? 

**Describe alternatives you've considered**
Anything I can generate ELF.

**Additional context**
Add any other context or graphics (drag-and-drop an image) about the feature request here.

nashif, 2020-03-11:
which board is that?

kkiyota, 2020-03-11:
@nashif It's my custom board.

kkiyota, 2020-03-12:
Any idea how I can minimize ELF file size?  Or move it to specific area?

nashif, 2024-02-08:
> Any idea how I can minimize ELF file size? Or move it to specific area?

Reduce the firmware size by removing features etc, it is very difficult to measure coverage on a production board if you enable everything, so start with the things interested in rather than include everything.

closed 2024-02-08 by nashif