zephyrproject-rtos/zephyr #23395: UART Console input does not work on SiFive HiFive1 on echo sample app

carlosedp, 2020-03-11:
**Describe the bug**

The UART (serial) console does not work on the SiFive HiFive1 board (tested on the physical board and [Renode Simulation Framework](https://renode.io)).

I've built the `echo` [sample](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/console/echo) application, loaded to the board and simulator and it does not print to the console by using the `console_write` or `console_putchar` commands.

I only see:

```
 You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
```

[image]

And nothing more. On Renode debugging console, whenever I press some key in the console, I see on the logs: `09:08:07.4953 [WARNING] uart0: Trying to read data from empty receive fifo` but the character is not echoed back.

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=hifive1 ../
3. make

The binary is correctly generated but presents the issue above.

To simulate the board, download and install Renode, build the ELF for the `echo` sample app. Open Renode and enter in the "Monitor" window (keep the @ characters):

```
include @scripts/single-node/sifive_fe310.resc
sysbus LoadELF @path/to/file/zephyr.elf
start
```

A new window will open with the serial console.

**Expected behavior**

Characters being correctly echoed back.

**Impact**

Nothing that requires serial console input works (notably, the shell).

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: MacOS Mojave. Using PlatformIO as development IDE.
 - Toolchain: Zephyr 2.1.0

**Additional context**
Add any other context about the problem here.

carlosedp, 2020-03-16:
Hi folks, any news on this?
Cc. @carlescufi @nategraff-sifive

nategraff-sifive, 2020-03-16:
I'm sheltering in-place for the foreseeable future and I don't have a board with me, so if you want progress from this it's not going to be from me, unfortunately. Sorry about that.

carlosedp, 2020-03-16:
Could you try with a simulated HiFive1 on Antmicro Renode?

Check: https://www.zephyrproject.org/developing-zephyr-rtos-embedded-applications-on-platformio-and-simulating-on-antmicro-renode/

carlescufi, 2020-03-23:
@carlosedp I suggest you  post in the mailing list and/or Slack to get more visibility. Aside from @nategraff-sifive, I know that there are a few ANT Micro employees that know this platform.

closed 2020-03-24 by nashif

carlosedp, 2020-03-24:
Confirming that both [echo](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/console/echo) and [shell](https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/subsys/shell/shell_module) sample applications worked on the Renode HiFive1 simulated board with this fix.
Thanks @pabigot!

PR #23699 (2020-03-24): uart: sifive: fix interrupt-driven transmission
A txcnt of zero prevents transmission, as transmit requires the number
of entries in the transmit fifo to be strictly less than the txcnt
value.  Set the default to 1.

Fixes #23395