zephyrproject-rtos/zephyr #23418: Building hello_world failed

swapnabulbule, 2020-03-12:
I have followed steps as per guid https://docs.zephyrproject.org/latest/getting_started/index.html#
Until step 6, building the application,receiving error.
OS: Ubuntu 18.04

1) export ZEPHYR_TOOLCHAIN_VARIANT=zephyr
export ZEPHYR_SDK_INSTALL_DIR=/home/swapna/Documents/Workspace/Zephyr_ESP

west build -p auto -b esp32 samples/hello_world
-- west build: build configuration:
       source directory: /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/samples/hello_world
       build directory: /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build (created)
       BOARD: esp32 (origin: command line)
-- west build: generating a build system
-- Zephyr version: 2.2.0-rc3
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6") 
-- Selected BOARD esp32
-- Found west: /home/swapna/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Loading /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts as base
Devicetree header saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
CMake Error at /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/cmake/extensions.cmake:1479 (message):
  No such file or directory: LIBGCC_FILE_NAME: ''
Call Stack (most recent call first):
  /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/cmake/compiler/gcc/target.cmake:69 (assert_exists)
  /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/cmake/target_toolchain.cmake:49 (include)
  /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/cmake/app/boilerplate.cmake:476 (include)
  CMakeLists.txt:5 (include)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/swapna/bin/cmake/cmake-3.16.4-Linux-x86_64/bin/cmake -B/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build -S/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/samples/hello_world -GNinja -DBOARD=esp32

I have tried couple of things changing variant and tool chain. 

2. export ZEPHYR_TOOLCHAIN_VARIANT=cross-compile
export CROSS_COMPILE=/usr/bin/arm-none-eabi-

I have followed steps as per guid https://docs.zephyrproject.org/latest/getting_started/index.html#
Until step 6, building the application,receiving error.

swapna@18:~/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr$ west build -p auto -b esp32 samples/hello_world
-- west build: build configuration:
       source directory: /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/samples/hello_world
       build directory: /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build (created)
       BOARD: esp32 (origin: command line)
-- west build: generating a build system
-- Zephyr version: 2.2.0-rc3
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.9", minimum required is "3.6")
-- Selected BOARD esp32
-- Found west: /home/swapna/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Loading /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts as base
Devicetree header saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 6.3.1
-- The CXX compiler identification is GNU 6.3.1
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- Cache files will be written to: /home/swapna/.cache/zephyr
CMake Warning at /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/common/esp32.board.cmake:7 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/boards/xtensa/esp32/board.cmake:3 (include)
[... 52 lines cut ...]
/tmp/ccTtYhaH.s:149: Error: junk at end of line, first unrecognized character is `1'
/tmp/ccTtYhaH.s:155: Error: bad expression
/tmp/ccTtYhaH.s:155: Error: junk at end of line, first unrecognized character is `1'
/tmp/ccTtYhaH.s:162: Error: bad expression
/tmp/ccTtYhaH.s:162: Error: junk at end of line, first unrecognized character is `4'
/tmp/ccTtYhaH.s:168: Error: bad expression
/tmp/ccTtYhaH.s:168: Error: junk at end of line, first unrecognized character is `0'
/tmp/ccTtYhaH.s:174: Error: bad expression
/tmp/ccTtYhaH.s:174: Error: junk at end of line, first unrecognized character is `8'
/tmp/ccTtYhaH.s:180: Error: bad expression
/tmp/ccTtYhaH.s:180: Error: junk at end of line, first unrecognized character is `0'
/tmp/ccTtYhaH.s:186: Error: bad expression
/tmp/ccTtYhaH.s:186: Error: junk at end of line, first unrecognized character is `0'
/tmp/ccTtYhaH.s:192: Error: bad expression
/tmp/ccTtYhaH.s:192: Error: junk at end of line, first unrecognized character is `0'
/tmp/ccTtYhaH.s:198: Error: bad expression
/tmp/ccTtYhaH.s:198: Error: junk at end of line, first unrecognized character is `4'
/tmp/ccTtYhaH.s:204: Error: bad expression
/tmp/ccTtYhaH.s:204: Error: junk at end of line, first unrecognized character is `8'
/tmp/ccTtYhaH.s:210: Error: bad expression
/tmp/ccTtYhaH.s:210: Error: junk at end of line, first unrecognized character is `9'
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /home/swapna/bin/cmake/cmake-3.16.4-Linux-x86_64/bin/cmake --build /home/swapna/Documents/Workspace/Zephyr_ESP/zephyrproject/zephyr/build

I am not sure which variant and toolchain

nashif, 2020-03-12:
```
west build -b esp32 samples/hello_world/
-- west build: generating a build system
-- Application: /home/nashif/Work/zephyrproject/zephyr/samples/hello_world
-- Zephyr version: 2.2.99
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.6", minimum required is "3.6")
-- Board: esp32
-- Found west: /home/nashif/.local/bin/west (found suitable version "0.7.2", minimum required is "0.6.0")
-- Found toolchain: espressif (/home/nashif/Work/SDK/xtensa-esp32-elf)
-- Loading /home/nashif/Work/zephyrproject/zephyr/boards/xtensa/esp32/esp32.dts as base
Devicetree header saved to '/home/nashif/Work/zephyrproject/zephyr/build/zephyr/include/generated/devicetree_unfixed.h'
Parsing /home/nashif/Work/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/nashif/Work/zephyrproject/zephyr/boards/xtensa/esp32/esp32_defconfig'
Merged configuration '/home/nashif/Work/zephyrproject/zephyr/samples/hello_world/prj.conf'
Configuration saved to '/home/nashif/Work/zephyrproject/zephyr/build/zephyr/.config'
Kconfig header saved to '/home/nashif/Work/zephyrproject/zephyr/build/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/nashif/Work/SDK/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Cache files will be written to: /home/nashif/.cache/zephyr
CMake Warning at /home/nashif/Work/zephyrproject/zephyr/boards/common/esp32.board.cmake:7 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/nashif/Work/zephyrproject/zephyr/boards/xtensa/esp32/board.cmake:3 (include)
  /home/nashif/Work/zephyrproject/zephyr/cmake/app/boilerplate.cmake:508 (include)
  CMakeLists.txt:5 (include)

-- Configuring done
-- Generating done
-- Build files have been written to: /home/nashif/Work/zephyrproject/zephyr/build
-- west build: building application
[1/122] Preparing syscall dependency handling

[122/122] Linking C executable zephyr/zephyr.elf
```

You are not using the right toolchain, please follow the documentation.

closed 2020-03-12 by nashif

swapnabulbule, 2020-03-12:
Can I know which toolchain are using?