zephyrproject-rtos/zephyr #23423: Watchdog: mitigation in case system [created] threads hang/non-responsive

n3rd4i, 2020-03-12:
**Describe the bug**
System [subsystem] created threads like: "BT RX pri" or "BT TX", etc, don't have a mitigation in place for checking "health" so that system could recover from that.

**To Reproduce**
see [23364](https://github.com/zephyrproject-rtos/zephyr/issues/23364)

**Expected behavior**
Ideally way to "recover" for "future" situations like this.

**Impact**
Could become showstopper if there is no way to recover.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: gnu-arm-toolchain
 - Commit 2.2.0

carlescufi, 2020-03-12:
The general solution here would be to "kill" unresponsive threads, but honestly I am not sure how that would improve the actual situation. Perhaps we could just reset the system in that case instead.

andrewboie, 2020-03-12:
> Ideally way to "recover" for "future" situations like this.

You need to be a lot more specific about what you want. As it stands this ticket is too vague.

n3rd4i, 2020-03-12:
I've updated the ticket.

Also `K_ESSENTIAL` cannot be set for these threads since they are defined by the OS. 

Could future releases include API to say: `k_thread_options_set` like `K_ESSENTIAL` on runtime?

andrewboie, 2020-03-12:
> Ideally, a way to "recover" trigger user-defined handler after detecting "future" situations non-healthy [system] threads [non-responsive or deadlocked] like 23364.

And how exactly does one detect this non-responsive / deadlocked situation? To me it seems there would need to be custom logic for every thread covered in this way.

joerchan, 2020-03-12:
I have never tried them, but there are watchdog drivers for many SoCs. Could they be used?

mfiumara, 2020-03-13:
> And how exactly does one detect this non-responsive / deadlocked situation? To me it seems there would need to be custom logic for every thread covered in this way.

It's not easy to detect non-responsiveness / deadlocks, however, we could deduct from some thread information in the application and act on it. Some thread info that is currently not available (as far as I know) that could be used for this is the amount of time a thread has been in it's current (blocking) state.
Another solution would be being able to providing a thread some kind of "hook" whenever it is scheduled. This would allow the user to track a thread's activity, then it's up to the user what to do with that information.

I'm not sure if there's anything like this already available in the kernel?

carlescufi, 2020-03-13:
> I have never tried them, but there are watchdog drivers for many SoCs. Could they be used?

A watchdog is global and will reset the whole system. I assume @n3rd4i wants a single thread or subsystem to be restarted, microkernel-style.

martinjaeger, 2020-05-07:
While trying to use the watchdog on an STM32, I encountered an issue that is related to this one.

As described in the documentation, `wdt_install_timeout` should return a channel ID which can then be used to feed the dog from multiple threads. However, only the NRF driver actually implements this feature. All other drivers just return 0 (or an error).

It seems like NRF supports multiple channels in hardware, so for other MCUs I suggest we implement the channel-handling in software (in the driver) to have a consistent behavior of the API and make the watchdog more useful for a multi-threaded environment. What do you think?

As an alternative, an own "kicker" thread for watchdog handling could be used, as described [here](https://stackoverflow.com/questions/13217959/how-to-use-the-watchdog-timer-in-a-rtos). But I think it would be better to handle it in the driver directly.

carlescufi, 2020-07-10:
The actual Bluetooth-related issue has been fixed. This issue is too vague, since many threads wait with `K_FOREVER` and there's nothing the kernel can do to recover from that if no other threads or ISRs wake it up. Also, resetting a thread would not be enough, since in many subsystems there are also other dependencies to separate threads.
I suggest we use a watchdog for now, or let the application have its own timer that can trigger a reset or, once it's a available, a bluetooth disable/enable cycle.

closed 2020-07-10 by carlescufi