zephyrproject-rtos/zephyr #23424: DTLS connection ID

endian-benjamin, 2020-03-12:
See the RFC at https://datatracker.ietf.org/doc/draft-ietf-tls-dtls-connection-id/

This seems like it would be really simple to implement in the TLS socket layer in Zephyr, basically just requires extending the `setsockopt` options. mbedTLS already supports it. Californium supports it on the server side.

Is this on the roadmap? Would be good to know before I start hacking on it myself.

rlubos, 2020-03-12:
Not on the roadmap and contributions are welcome :) It sounds reasonable to allow Connection ID to be configurable as a secure socket option.

One note though, it seems to be supported since mbedTLS 2.18.0, in Zephyr we currently have mbedTLS 2.16.4. So mbedTLS version bump would be needed first.

endian-benjamin, 2020-03-13:
@rlubos And the fact that 2.18 is not LTS would be okay? (not even sure they plan on releasing any more 2.x LTS:es)

endian-benjamin, 2020-03-13:
Oops

jukkar, 2020-03-14:
> And the fact that 2.18 is not LTS would be okay?

I think we can update the mbedTLS as needed.

endian-benjamin, 2020-05-29:
How do I talk to regarding updating mbedTLS? `CODEOWNERS` isn't clear. @nashif ? @ceolin ?

ceolin, 2020-05-29:
It is not clear for me if we will have another 2.x LTS release and from 3.0 release mbedTLS will be splitted into two, crypto and TLS. We will need to come up with a plan to support this.
Until there I think it is problematic  uses a version that will no be supported and not receive the proper fixes.

endian-benjamin, 2020-06-02:
@jukkar seems to think LTS is not 100% mandatory.

Note though that IMO connection ID is a pretty killer feature for IoT, based on discussions with various industry players. It's not some fringe nice-to-have.

jukkar, 2020-06-02:
> @jukkar seems to think LTS is not 100% mandatory.

What do you mean by this, I am not  following you?

endian-benjamin, 2020-06-02:
@jukkar https://github.com/zephyrproject-rtos/zephyr/issues/23424#issuecomment-599046472

jukkar, 2020-06-02:
> @jukkar [#23424 (comment)](https://github.com/zephyrproject-rtos/zephyr/issues/23424#issuecomment-599046472)

Yes, I got that reference, I just do not understand you comment. How my comment "I think we can update the mbedTLS as needed." is related to to your comment. All I am saying that if there are bugs in zephyr used mbedtls, we can always upgrade to newer version, it was just a generic comment. How mbedtls LTS release is relevant here?

boaks, 2021-01-24:
That would be a very cool extension feature.

I use DTLS Connection ID now for about 2 years with CAT-NB/NB-IoT modems (Raspi-java-Californium-at-cmds-serial-modem-client) and in my experience, this is the only technique,which works without too much headaches. It also overcomes some k8s networking obstacles, so it's not only for NB-IoT.   

Remarks:

mbedtls is unfortunately not using the IANA values (see [Please adjust MBEDTLS_TLS_EXT_CID to 53](https://github.com/ARMmbed/mbedtls/issues/3892) ). Unfortunately, I couldn't convince the mbedtls team to adjust the value nor to allow to specify that externally. Anyway, if clients want to connect to Eclipse/Californium (or Eclipse/Leshan), you need either to patch mbedtls or Californium to use the same value for the hello extension.

The IETF TLS  working group has changed the MAC calculation (see [Update to new (hopefully) injective MAC structure as discussed in the meeting and on-list](https://github.com/tlswg/dtls-conn-id/pull/77) ). AFAIK mbedtls hasn't adjusted the MAC yet, nor do I plan to adjust it in Californium in the next time. (It's no big task to adjust it, once the RFC gets released or mbedtls updates their implementation). 

That makes it more a pioneers feature and requires an explicit statement, that there may be "breaking changes" until the RFC is released. Anyway, I think, that gathering experience with it, it's worth doing it now. Using a patched mbedtls version (IANA extension value 53), I already connected libcoap-mbedtls to Californium, so it should work without too much trouble.

rlubos, 2021-07-05:
I've submitted a draft PR with DTLS Connection ID socket option, see https://github.com/zephyrproject-rtos/zephyr/pull/36738. I think however that we should postpone the integration of this feature until some problems on the mbed TLS side are resolved (see the PR, where I've described in a more detail the issues I've faced).

@boaks Thank you for all the feedback that you've provided across different repositories (Zephyr/mbedTLS/Leshan). It helped a lot to investigate the initial failures I've encountered when testing the feature.

boaks, 2022-03-22:
Just to mention:
[RFC9146](https://www.rfc-editor.org/rfc/rfc9146.html) is published.

boaks, 2022-05-09:
See [zephyr/mbedtls](https://github.com/zephyrproject-rtos/mbedtls/pull/36)

boaks, 2022-11-29:
Some latest news:

The new https://github.com/Mbed-TLS/mbedtls/pull/6264 has been merged.

motters, 2023-06-05:
Little nudge on this / https://github.com/zephyrproject-rtos/zephyr/pull/36738.

Side effect of not having this is poor performing LwM2M. Due to the mobile network changing the port / ip.

When lifetimes over 5 minutes are used on cellular networks update requests always fail. Forcing the LwM2M stack to wait for a timeout before attempting a full-registration (increased data usage + device being marked as offline). It also means a high likelihood of LwM2M send requests failing.

hasheddan, 2023-07-17:
Hey folks! I'd be interested in working on this if it has fallen out of priority for others -- @rlubos @boaks is there any ongoing work I should know about? Is there a desire to revive https://github.com/zephyrproject-rtos/zephyr/pull/36738?

boaks, 2023-07-17:
AFAIK:
mbedTLS 3.3.0 (Dec. 2022) has now a RFC 9146 compliant implementation.
For the nRF9160 this is in the meantime also part of the mfw 1.3.5.

I don't know, if it's possible to use the zephyr/mbedTLS stack with enabled DTLS 1.2 CID support,
but maybe Robert knows that.

(I still use the tinydtls cid feature branch.)

rlubos, 2023-07-18:
> is there any ongoing work I should know about?

Not that I'm aware of. There was also an attempt in https://github.com/zephyrproject-rtos/zephyr/pull/44914, built on top of https://github.com/zephyrproject-rtos/zephyr/pull/36738, which included extra cherry-picks for mbed TLS, but it also did not make it. Now that we have a proper mbed TLS version in Zephyr we could revive the attempt, I unfortunately didn't have a time to do that. So if you're willing to contribute, feel free to do so.

Now the question is how do we want to support CID option. In the first prototype I proposed that the option accepts an array representing the CID to use, so the responsibility to generate CID is on the application. The aforementioned nRF9160 sockets took a different approach, where the CID option is simply used to enable/disable CID support, and the underlying implementation handles all of the details. Maybe that would be easier for the applications to use. I'm open for discussion here.

boaks, 2023-07-18:
Though I'm more used to my own implementations (tinydtls and Californium), I just tried out the nRF9160 mfw 1.3.5 variant right now.

[networking sockets - NRF_SO_SEC_DTLS_CID](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrfxlib/nrf_modem/doc/sockets.html)

with 

```
NRF_SO_SEC_DTLS_CID

    Set DTLS Connection ID setting. The socket option is supported from modem firmware v1.3.x, where x is greater than or equal to 5, and v2.x.x. The following values are accepted:

        NRF_SO_SEC_DTLS_CID_DISABLED - The DTLS connection ID is not used, and the connection ID extension is not included in the client hello.

        NRF_SO_SEC_DTLS_CID_SUPPORTED - The DTLS connection ID is supported but not used, and the connection ID extension with a zero-length CID is included in the client hello.

        NRF_SO_SEC_DTLS_CID_ENABLED - The DTLS connection ID is used, and the connection ID extension with a valid CID is included in the client hello.

    The default is disabled.
```

That's similar to the configuration API I introduced some years ago in Californium. AFAIK, the only difference is, that in Californium I can configure the length of the CID, while the modem seems to use 8 bytes. Anyway, pretty easy to use, just additional:

```
	int cid = NRF_SO_SEC_DTLS_CID_ENABLED;
	err = setsockopt(sock, SOL_TLS, TLS_DTLS_CID, &cid, sizeof(cid));
```

hasheddan, 2023-07-18:
> So if you're willing to contribute, feel free to do so.

@rlubos thanks for the context -- I would love to!

> Now the question is how do we want to support CID option. In the first prototype I proposed that the option accepts an array representing the CID to use, so the responsibility to generate CID is on [...]

I generally would lean towards allowing configuration at the application level. I tested out CID support in mbedTLS yesterday, and it allows for setting a variable length CID.

```c
/**
 * \brief             Configure the use of the Connection ID (CID)
 *                    extension in the next handshake.
 *
 *                    Reference: RFC 9146 (or draft-ietf-tls-dtls-connection-id-05
 *                    https://tools.ietf.org/html/draft-ietf-tls-dtls-connection-id-05
 *                    for legacy version)
 *
 *                    The DTLS CID extension allows the reliable association of
 *                    DTLS records to DTLS connections across changes in the
 *                    underlying transport (changed IP and Port metadata) by
 *                    adding explicit connection identifiers (CIDs) to the
 *                    headers of encrypted DTLS records. The desired CIDs are
 *                    configured by the application layer and are exchanged in
 *                    new `ClientHello` / `ServerHello` extensions during the
 *                    handshake, where each side indicates the CID it wants the
 *                    peer to use when writing encrypted messages. The CIDs are
 *                    put to use once records get encrypted: the stack discards
 *                    any incoming records that don't include the configured CID
 *                    in their header, and adds the peer's requested CID to the
 *                    headers of outgoing messages.
 *
 *                    This API enables or disables the use of the CID extension
 *                    in the next handshake and sets the value of the CID to
 *                    be used for incoming messages.
 *
 * \param ssl         The SSL context to configure. This must be initialized.
 * \param enable      This value determines whether the CID extension should
 *                    be used or not. Possible values are:
 *                    - MBEDTLS_SSL_CID_ENABLED to enable the use of the CID.
 *                    - MBEDTLS_SSL_CID_DISABLED (default) to disable the use
 *                      of the CID.
 * \param own_cid     The address of the readable buffer holding the CID we want
 *                    the peer to use when sending encrypted messages to us.
 *                    This may be \c NULL if \p own_cid_len is \c 0.
 *                    This parameter is unused if \p enabled is set to
 *                    MBEDTLS_SSL_CID_DISABLED.
 * \param own_cid_len The length of \p own_cid.
 *                    This parameter is unused if \p enabled is set to
 *                    MBEDTLS_SSL_CID_DISABLED.
 *
 * \note              The value of \p own_cid_len must match the value of the
 *                    \c len parameter passed to mbedtls_ssl_conf_cid()
 *                    when configuring the ::mbedtls_ssl_config that \p ssl
 *                    is bound to.
 *
 * \note              This CID configuration applies to subsequent handshakes
 *                    performed on the SSL context \p ssl, but does not trigger
 *                    one. You still have to call `mbedtls_ssl_handshake()`
 *                    (for the initial handshake) or `mbedtls_ssl_renegotiate()`
 *                    (for a renegotiation handshake) explicitly after a
 *                    successful call to this function to run the handshake.
 *
 * \note              This call cannot guarantee that the use of the CID
 *                    will be successfully negotiated in the next handshake,
 *                    because the peer might not support it. Specifically:
 *                    - On the Client, enabling the use of the CID through
 *                      this call implies that the `ClientHello` in the next
 *                      handshake will include the CID extension, thereby
 *                      offering the use of the CID to the server. Only if
 *                      the `ServerHello` contains the CID extension, too,
[... 21 lines cut ...]
 * \return            A negative error code on failure.
 */
int mbedtls_ssl_set_cid(mbedtls_ssl_context *ssl,
                        int enable,
                        unsigned char const *own_cid,
                        size_t own_cid_len);

```

Mapping to the nRF example @boaks provided above, it looks like:

CID disabled (i.e. `NRF_SO_SEC_DTLS_CID_DISABLED`):
```c
mbedtls_ssl_set_cid(&ssl, MBEDTLS_SSL_CID_DISABLED, NULL, 0);
```

CID supported (i.e. `NRF_SO_SEC_DTLS_CID_SUPPORTED`):
```c
mbedtls_ssl_set_cid(&ssl, MBEDTLS_SSL_CID_ENABLED, NULL, 0);
```

CID enabled (i.e. `NRF_SO_SEC_DTLS_CID_ENABLED`):
```c
mbedtls_ssl_set_cid(&ssl, MBEDTLS_SSL_CID_ENABLED, own_cid, sizeof(own_cid));
```

plskeggs, 2023-07-18:
When using a DTLS connection as a client, it is not always necessary for the client to provide a CID, but only to indicate in the handshake to the server that it supports CID by providing a zero length CID.  A server which implements CID will respond with a CID, which the client then needs to present each time it initiates communication with the server.

Apparently the converse can also be true.

In that light, it is not necessary to always provide a non-zero-length CID as a client, so if not set, there should be a way to not provide a default one under the hood.

hasheddan, 2023-07-18:
> In that light, it is not necessary to always provide a non-zero-length CID as a client, so if not set, there should be a way to not provide a default one under the hood.

@plskeggs agreed -- I believe the mode you are referring to matches the following:

> CID supported (i.e. NRF_SO_SEC_DTLS_CID_SUPPORTED):
```c
mbedtls_ssl_set_cid(&ssl, MBEDTLS_SSL_CID_ENABLED, NULL, 0);
```

The server may also opt to support sending CID but not require that the client send one, though that use case seems to be much less common.

boaks, 2023-07-19:
> When using a DTLS connection as a client, it is not always necessary for the client to provide a CID, but only to indicate in the handshake to the server that it supports CID by providing a zero length CID. 

That was the intention when I designed that configuration API for Californium. In the most common use-case, the server use a static IP-address while the client will be NATed. The client is therefore free to decide to go with supported (means, the client will send the cid, assigned by the server, in its records to the server) or enabled (means, both are using the cid, assigned by the other peer, in their records). 

In the past years I'm only aware of some rare cases (CID load balancer), where it makes sense to use a other value for the cid then just random bytes. In Californium I added therefore callbacks. For a client, I'm not sure, if that makes really sense. In the very most use-cases, just providing a proper cid length will do it.

plskeggs, 2023-08-30:
PR: https://github.com/zephyrproject-rtos/zephyr/pull/61483 merged.  I think this issue can be closed now.

closed 2023-08-31 by rlubos