zephyrproject-rtos/zephyr #23425: Remote opencd

steros76, 2020-03-12:
I use Raspberry PI as programmer for NRF 52832 chips.
Now I installed Zephyr on my PC and switched my project to this platform.
I see openocd is supported by Zephyr but it seems to me only for local use.
Is there a way to automate deployment with west tool using opencd listening on remote 4444 port of my Raspberry?
Thanks.

mbolivar, 2020-03-13:
I don't know what this would take, sorry. I personally do not use openocd very much. Can you explain what you would do with openocd to get this done without using west?

mbolivar-nordic, 2020-03-17:
I've switched my work-related account to @mbolivar-nordic (https://github.com/zephyrproject-rtos/zephyr/pull/23530).

steros76, 2020-03-19:
Sorry, at the moment I can't give you exact command to execute.
Using command line I do it in 2 steps but using Eclipse it works automatically in remote mode.
I should investigate deeply how it works behind the scene. I'll post the instructions when I understand it. Sorry for the delay.

mbolivar-nordic, 2020-03-19:
There's no rush. Thank you for your response.

carlescufi, 2020-04-07:
We have no plans to implement this functionality. Please send a Pull Request if you have interest in this becoming part of the tree.

closed 2020-04-07 by carlescufi