zephyrproject-rtos/zephyr #23438: Cannot reset Bluetooth mesh device

johnsmithecon, 2020-03-13:
**Describe the bug**
I am only able to reset the bluetooth mesh device by calling bt_mesh_reset(); in the main() function right after calling bt_ready();
This means that when I try to reset the device by calling bt_mesh_reset(); from a callback function for example that is not in main.c, the node does not reset. I can verify this by provisioning a node using a bluetooth mesh app then scanning for bluetooth mesh devices (I don't see the device because it is provisioned). After trying to reset from my callback function, my node is still not seen on the scan meaning that it did not fully reset. I also try power cycling and the node is still unscannable.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a Bluetooth mesh sample project (I used the onoff_level_lighting_vnd_app one for nrf52_pca10040)
2. enable gpio button
3. call bt_mesh_reset() reset when gpio button is triggered

**Expected behavior**
Node should reset

**Impact**
What impact does this issue have on your progress (e.g., annoyance, showstopper)

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

jhedberg, 2020-03-13:
From the Slack discussion it sounded like you're using timer callbacks (i.e. k_timer). Those callbacks are in interrupt context, which I'm not surprised doesn't work well together with bt_mesh_reset(). I'd suggest you instead use the k_delayed_work API so that your callback is scheduled in thread context rather than interrupt.

johnsmithecon, 2020-03-13:
But I tried to call bt_mesh_reset() from button handler and it didn't do anything. I'm trying to create a reset scenario where user presses and holds button and after 3 seconds of holding, the device resets. I am doing this by adding 3 second timer on button press and the timer callback reads the gpio pin of the button and determines if it is high. If it is high then I call bt_mesh_reset(). If I can't even call bt_mesh_reset() during button press callback then what should I do?

johnsmithecon, 2020-03-13:
Update: Your solution works. Thanks so much for the help and the quickness of the responses.

closed 2020-03-13 by jhedberg